MarlinFirmware/Marlin #15051: [BUG] (marlin 2.0 Status screen error

cinealfa, 2019-08-24:
### Description

Will not compile
### Steps to Reproduce
Use the latest branch transfer all settings to new config

1.Compile

**Actual behavior:** [What actually happens]
Error
[image]

#### Additional Information
both the stock status _screen and -bootscreen files are fresh and in the right directory( I made that mistake already).
tried turning the custom screens off- no change
changed any setting with status in it or encoder counts still no change.
I am assuming it is transcription error on my part but I cannot seem to find it.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3537613/Archive.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AnHardt, 2019-08-24:
Don't use `MARLIN_DEV_MODE` with a LPC1768. It will only work when user on an AVR processors.
Don't activate random config options, when you don't know what they are for.
Never change more than one option at a time, else you cant keep track on what caused errors. (Or you have to be able to interpret the code and the errors.)

cinealfa, 2019-08-25:
Thank you for your solution @AnHardt You caught it very quickly!  (albeit with a bit of a patronizing tone.) I am well aware that the changing of more than one parameter at a time is a recipe for confusion. As I said, I figured it was transcription error over to a new set of config files. And in fact,that is what it was. I know what TCHT5 is. WHich only confused matters more as I am not in the Arduino realm. One wonders should there be a sanity check for the dev mode? I would have save  me hours and you having to /HELPscold me ;)

closed 2019-08-25 by cinealfa