MarlinFirmware/Marlin #15052: [FR] Allow save the printer settings into a gcode file

sl1pkn07, 2019-08-25:
### Description

Since all the settings is configurable via gcode (z probe offsets, tmc configurations, jerks/acceleration, etc), is great if that configurations can be stored in the media (if is available) in a gcode file

advantages over the eeprom is, if you reset the eeprom or upgrade the firmware to new version, the configurations they get lost. then you only need "print" the gcode for restored the configurations

greetings

AnHardt, 2019-08-26:
We already produce a text file - on the serial console. You just have to copy in a file and insert some ";".
I have a whole bunch of different configs for different purposes on SD.
My default config is named `auto0.g`.

AnHardt, 2019-08-26:
Just M115 is not a readeble config. It's for informing the host program about what the firmware can do.
Better try M503.

MarkMan0, 2019-08-29:
Are the data from [M503(Report settings)](http://marlinfw.org/docs/gcode/M503.html) and [G29 S-1(UBL mesh)](http://marlinfw.org/docs/gcode/G029-ubl.html) sufficient? 
I mean, if we format the output of these two gcodes, we can generate a script, which sets up the machine, as if it read everything from the eeprom right?
Or did I miss something?

AnHardt, 2019-08-29:
Both of the commands are mend for exactly that purpose. They should output all information that is stored in EEPORM in a human and machine readable manner (g-code).  If not that is a bug and will be corrected.

sl1pkn07, 2019-08-30:
then, now the "error"/FR is for is not able to save directly in a gcode file (in the SD), directly from LCD, and without "external" intervention (pronterface/octoprint/etc)

Vertabreak, 2019-11-13:
octoprint also has a eeprom dump plugin.

sl1pkn07, 2020-07-01:
The petition of this FR is still up

thinkyhead, 2020-07-03:
The output of `M503` can be copied into a G-code file, such as "`auto0.g`". Marlin doesn't have the methods available to capture `M503` output to a file from within the firmware, and this low-level hook is not likely to be added anytime soon. To fill in for this architectural deficit, you can use the EEPROM and plugins like the one mentioned above.

closed 2020-07-03 by thinkyhead