MarlinFirmware/Marlin #15053: [BUG] autolevel fade height and extruder "output"

spilz87, 2019-08-25:
### Description

When I use fade height, Z move when X and Y move, which is normal, but Extruder put the same quantity of material what ever if fade "increases" or "reduces" the actual layer 

### Steps to Reproduce

use a non perfect leveled bed
add G29 and enable 1mm fade height  
-> where bed is lower, there is not enouth filament
-> where bed is higher, there is too much filament

I expect fade height change the extruder "output" according to Z compensation

regards

AnHardt, 2019-08-25:
Set a larger fade height to make the difference between the layers smaller. Than the different worm shape (more or less squeezed) is able to bridge the vertical gaps (on cost of width). 

"Extrusion Amount Leveling" still isn't implemented in Marlin. Is it somewhere else?

spilz87, 2019-08-25:
Thanks for your reply
It’s a way to compensate it 

Regards

Roxy-3D, 2019-08-25:
1 mm of 'Fade Height' is not very much.   There is a reason 10 mm is the default amount.

With that said...   the math to adjust the extrusion amount is not very complicated.  I'm pretty sure we don't want to add any trigonometric calculations into the calculations.   But we can probably get very good approximations on how much to scale the number based on the length of the line segment and the difference in Z-Height correction.

@AnHardt   Do you agree that we could get a good approximation on the scaling factor based on the line segment length and the difference in Z-Height correction?

AnHardt, 2019-08-25:
Never thought about that. Always planed to try to level, by extrusion amount, only at the first layer. But never found the time.

AnHardt, 2019-08-25:
If you have the sliced extrusion width the additional filament volume is easily calculated from the difference between ideal and measured height. Together with the planed layer height a % or factor  value could be calculated.

Roxy-3D, 2019-08-25:
Probably the biggest complication to adding the extrusion amount to the Z-Height correction is this:

The way the Z-Height correction is done, the desired Z height doesn't really change.   The firmware is just adding or subtracting a little bit to that amount.    That strategy doesn't work on the E-Axis because the machine is constantly pushing filament through the nozzle.   So, the firmware can't just decide to add a little bit more and bump the number up.   

Instead, we would have to schedule the E-Axis moves with a little bit more (or less) length and then correct the 'current position' to reflect where it would have been without any correction.

AnHardt, 2019-08-25:
If we divide the fade-height by the layer-height we get the amount of the above value to correct per layer.

Roxy-3D, 2019-08-25:
> If you have the sliced extrusion width the additional filament volume is easily calculated from the difference between ideal and measured height. Together with the planed layer height a % or factor value could be calculated.

Agreed...   But we don't really know the layer height.   However...  I don't think that matters.   The layer height is factored into what the slicer produces for the extrusion amount.   So if we just scale up (or down) what ever the GCode says to extrude for the line segment...   I think the layer height pretty much drops out of the equation.

AnHardt, 2019-08-25:
How to get the numbers is a different theme and probably will be different if we are in `volumetric` mode or not.

Roxy-3D, 2019-08-25:
Yes.   Good Point.

boelle, 2019-09-24:
@spilz87 still having problems?

spilz87, 2019-09-24:
It seems to change from a bug to a request but we can close as it’s not plane to add it

boelle, 2019-09-24:
i have changed the title and added the FR label and will reopen, some day maybe