MarlinFirmware/Marlin #15056: PID bug?

Tinchus2009, 2019-08-26:
Marlin version 2.0.x

Hello, just wanted to report this:
I was previously useing marlin 1.1.9 with great results. My PID settings were working really good with a temperatures with a variation of only +-0.3 degrees

I wanted to check the new features in 2.0.x so I moved there. 
I copied thr same PID settings, even the values. But now the temperature variations are too big. I reach target temp and them temperature drops like 7 dregrees, then goes again to target, sometime it overshoots 1 or 2 degrees, then drops 7 degrees or 8.
I ran the pid auto temp again, but I got the same results.

I went back to marlin 1.1.9 and everything works good again.
I have atached my configuration.h just in case you want to take a look and discard a confifuration problem before checking a posible bug.
My hardware is a arduina mega 2560 board.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3538909/Marlin.zip)

Grogyan, 2019-08-26:
Did you update your eeprom settings when you switched from 1.1.9 to 2.0?

Tinchus2009, 2019-08-26:
Yes, I erased the eprom completly just to be sure

GMagician, 2019-08-27:
PID handling code has been changed in 2.0, recently. Not sure old values are still valid. try to re-execute tuning

AnHardt, 2019-08-27:
Do you talk about the bed or the extruder?
If bed - PID is not on.
If extruder - why SLOW_PWM_HEATERS?

blueeagle69, 2019-08-28:
I have had the same issue since installing my SKR 1.3 and Marlin 2.0.
I have done multiple PID tuning runs, makes no difference.
Originally my extruder temp was overshooting by around 13 degrees on 2.0. With 1.9 it was perfect.
I have managed to get it down to an overshoot of around 7 degrees by playing about with PID_FUNCTIONAL_RANGE. Much better now, but slows down the print start or when changing filament etc, because it overshoots and then has to cool down to it's proper temperature.

BrokenFish, 2019-08-30:
I have the same problem on my skr v1.3. Sometimes when i preheat it is perfect afterwards. But the moment I start the print I get big variations. [https://photos.app.goo.gl/ZZh5NYtWm1MoKiTD9](url)

boelle, 2019-09-24:
@Tinchus2009 is this still a problem?

Tinchus2009, 2019-09-25:
Yes. I tried everything. I have to say that is not the first time I use Marlin, Im using it and configuring printer with it sine at least 3 years, so Im quit sure Im not making a config mistake. Also I know how to set up a PID, even manually. I have tried everything, and it doesnt work. I went back not to marlin 1.1.9 and PID is working perfectly, so I can discard hardware issues too. Something is not well with PID in version 2 when used in 8 bit electronics.

boelle, 2019-10-21:
how did you "Yes, I erased the eprom completly just to be sure"  ?

thinkyhead, 2019-10-21:
Sometimes it can help to just manually tune the PID. If you can find a range of values that work well, we can compare those to the PID values that worked with 1.1.9 and see if there's some correlation or pattern to the discrepancy. So far I have not experienced PID problems, but I also haven't been paying the closest attention to my temperature graphs.

Tinchus2009, 2019-10-22:
> 
> 
> how did you "Yes, I erased the eprom completly just to be sure" ?

From arduino IDE, there is an example script that erases the eprom.

boelle, 2019-10-22:
that one will do, but also you can just send M502 and then M500

boelle, 2019-10-24:
@Tinchus2009 any news?

Tinchus2009, 2019-10-24:
No, I gave up with version 2.X, I went back to 1.1.9

boelle, 2019-10-30:
will vlose this one

we can always reopen if multiple users can confirm there is an issue and we have their configs

closed 2019-10-30 by boelle