MarlinFirmware/Marlin #15062: [FR] (feature request title)

manarsak, 2019-08-26:
Hi I have an mini ups when I want to lose power  that moves the axis Z 2mm
[image]

boelle, 2019-08-26:
i assume that you are asking how to use the mini ups? its hard to guess as you are only telling you have one

manarsak, 2019-08-26:
I think you did not understand what I mean I mean when the loss of energy moving axis in order not to spoil the printing I have made several attempts but I face some mistakes

boelle, 2019-08-26:
attach your 2 config files as a zip, some smarter than me might be able to figure out if you have it configured wrong

manarsak, 2019-08-26:
Marlin.cpp

#if ENABLED(POWER_LOSS_RECOVERY)
void powerFailedDetect(void) {
  static bool isPowerFailed = false;
  if (!isPowerFailed && READ(POWER_LOSS_PIN) == POWER_LOSS_STATE) {
    isPowerFailed = true;
    thermalManager.disable_all_heaters(); //turn off heaters
    disable_all_steppers();

    SERIAL_ERROR_MSG("Power failed!");
      queue.clear();
    quickstop_stepper();

   queue.enqueue_now_P(PSTR("G91\nG1 E-10\nG1 Z10\n"));
    while (1) {
    #if ENABLED(USE_WATCHDOG)
      watchdog_reset();
    #endif
     queue.advance();   
      manage_inactivity(
        #if ENABLED(ADVANCED_PAUSE_FEATURE)
          no_stepper_sleep
        #endif
      );
    }
  }
}
#endif

#if !ENABLED(POWER_LOSS_RECOVERY)
static bool power_detect_enabled = false;
void GcodeSuite::M413()
{
  if (parser.seen('S'))
    power_detect_enabled = parser.value_bool();
  else {
    SERIAL_ECHO_START();
    SERIAL_ECHOPGM("Power-loss recovery ");
    serialprintln_onoff(power_detect_enabled);
  }
}              
#endif

AnHardt, 2019-08-26:
> Zaxis should NOT be moved on power loss. It is expected that it will retain last known position by resting on the print. If it has raised, it will most likely drift down slightly with no power and the risk of the print being scrapped instead of just a possible bump if stopped on a perimeter is extremely high. X/Y can rehome but Z cannot.

Bla, bla, bla! Lift and home to z-max.

AnHardt, 2019-08-26:
```
disable_all_steppers();

SERIAL_ERROR_MSG("Power failed!");
  queue.clear();
quickstop_stepper();
queue.enqueue_now_P(PSTR("G91\nG1 E-10\nG1 Z10\n"));
```
Doesn't make much sense. Disabling the steppers before the relative retraction and lift.

manarsak, 2019-08-27:
I want to lift in order not to spoil the printing

manarsak, 2019-08-27:
When we resume printing, we can reduce the axis with the same value we raised or can we move the axis X

manarsak, 2019-08-27:
Thank you so much. Can I ask you another question

manarsak, 2019-08-27:
I'm not stubborn. I just want to take advantage of this feature just to change. Thank you Marlin

manarsak, 2019-08-27:
Sir, I can when the loss of energy to save the place and then move the axis a little This panel is enough to move the axis when the loss of energy I want that when the loss of energy to save the current place and then move the axis step up or we can move the X axis instead of Z
We can shoot a video to see how the axis moves when energy is lost

manarsak, 2019-08-27:
Thank you I have solved the problem, which is when the loss of energy saves the place and then move the X axis to the beginning and after the resumption of printing returns X converter to the place before the loss of energy Thank you very much

AnHardt, 2019-08-27:
Going exactly back to the stored position is perfectly possible. Alone the stored position is not the position the printhead had when stopping the print. The planner buffer makes the difference. That's rarely observed because in the perimeters the way in the buffer is usually short and in the infill, where the ways are long, the error is covered by the next layer. But if a layer change is lost in the buffer ...

Grogyan, 2019-08-27:
@manarsak The mini UPS is kinda a trap.

It will not do, what you think it does, and the previous responses are accurate.

I will however say that, the mini UPS will NOT provide enough power to do anything like moving an axis on power loss. Josef Prusa has said, with his machines, and I strongly suggest looking into such a low cost Power Loss Recovery system.
Prusa Research developed Power Panic, to detect on the mains input, as soon as power is lost, thus using the really, big, capacitors in the power supply to put the machine into a safe state.
Another alternative to the Power Panic, is the MKS DET board, which operates in a similar fashion.