MarlinFirmware/Marlin #15067: [FR] BLTouch accidental pin drop prevention option

creamywhite, 2019-08-26:
Would it be possible to add an option for BLTouch-style ABL sensors to keep the coil powered if the last command was "stow pin"? My BLTouch (actually, a clone - 3DTouch) sometimes drops the pin during printing (for yet an unknown reason) after I switched from Marlin 1.1.9 to Marlin 2.0. If the coil will stay powered it will prevent accidental pin drops and stow the pin back up right away even if it manages to drop accidentaly.

creamywhite, 2019-08-28:
> Are you using the most current version of Marlin 2.0? The reason I'm asking is because I had the exact same problem as you (I'm using a genuine BLTouch though), and it was due to some changes made with the v3.0 BLTouch probes. The version of Marlin that released around the same time that the BLTouch v3.0 released didn't have the proper settings configured, and it was causing the new BLTouch probes to drop the pin when it shouldn't.
> 
> Older versions of Marlin (I believe around April) didn't have all the settings properly configured, which lead to a very weak hold on the BLTouch pin and the pin would very frequently drop during probes and regular printing. However, once the proper settings got integrated into the release, it seems to have fixed it. I haven't had that problem since, now the pin is dropped and retracted very solidly and there's no accidental deployments.

I'm using about a month-old version of Marlin 2.0. And I don't experience frequent drops. During probing even 5x5 grid it works flawlessly, and it doesn't drop during every single print (I'm using same exact gcode file to test).

Vertabreak, 2020-01-14:
Only time I experienced pin drop was when my wire harness had been breaking due to wire stress. when I moved the wires it would lose connection causing it to drop and reset then enter error state.

creamywhite, 2020-01-14:
It became evident that the board itself (SKR Mini E3) was causing the issue. Soldering small (100 nF) decoupling capacitor between Signal and Ground pins of probe connector solved the issue.

closed 2020-01-14 by creamywhite