MarlinFirmware/Marlin #15073: [BUG / FR / RFE] Marlin to include both machine and work coordinates in their position reports

telleropnul, 2019-08-27:
The following information is taken from https://github.com/cncjs/cncjs/issues/514

> Other controllers have a soft-configurable (does not require recompilation) option to include both machine and work coordinates in their position reports. For example, g2core tags machine coordinates with "mpos:" and work coordinates with "wpos:" in the JSON-formatted reports.

>In the case of Marlin, there is no easy way to coerce it to automatically issue timely position reports. You have to send a command asking it for a report, and the periodic sending of those commands sometimes causes motion stuttering because the "send a report" command has to be interspersed with the GCode from the file.

>The lack of automatic position reports can be troublesome for milling, where it is common for a single GCode move to take several seconds - for example a long arc at a low feedrate. With Marlin, the DROs don't update during that move, and might not update until the end of several such moves, depending on how often CNCjs has inserted "send position" commands.

AnHardt, 2019-08-29:
The M114 output is read from many different host programs.
Changing the output format or meaning is almost impossible. All hosts would have to follow the change.
Maybe a additional parameter to M114 could work, giving the new output only when requested with that parameter. Could also switch the format of the auto-reports.

telleropnul, 2019-09-03:
Where in my post did I say M114?  This is not about M114.

This request requires a Marlin dev with a bit of a CNC background.  I am greatful Marlin offers a CNC coordinate system and hope Marlin continues to evolve supporting CNC machine use.  If so, at some point the issues raised may need to be addressed.

Domush, 2022-02-07:
I've added support for this: https://github.com/MarlinFirmware/Marlin/pull/23689

```
#define M114_RAPID_REPORTING    // Outputs M114 position report to host after every G0 (rapid) move (for faster host UI updates), requires G0_FEEDRATE

/**
 * CNC Coordinate Systems
 *
 * Enables G53 and G54-G59.3 commands to select coordinate systems
 * and G92.1 to reset the workspace to native machine space.
 */
#define CNC_COORDINATE_SYSTEMS

#if ENABLED(CNC_COORDINATE_SYSTEMS)
  #define REPORT_MACHINE_POSITION // Returns both Work and Machine coordinates in default M114 report, similar to GRBL.
#endif
```