MarlinFirmware/Marlin #15077: Marlin2.0 case light flickering

McPaul0, 2019-08-27:
Hi!

I've just completed my case light. The LED is connected with the FET to some of available pins on the SKR V1.3 board.
When I set the brighness below 255 - it flickers.
#define fast_PWM - solves that problen, but the part cooling fan doesn't run on any speed below 255.

The pins I've tried 3.25 1.22 2.0 all the same issue.

Marlin 1.1.9 has no that problem on MKS Gen L board.

boelle, 2019-09-24:
@McPaul0 is the problem still present?

McPaul0, 2019-09-25:
I've just solve it. 
In configuration_adv.h
#if ENABLED(FAST_PWM_FAN)
  #define FAST_PWM_FAN_FREQUENCY 977

Is it correct solution?

boelle, 2019-09-25:
if the fan also work fine then yes

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle