MarlinFirmware/Marlin #15079: [FR] STM32F4 SDIO Support

ASPtr, 2019-08-28:
Is SDIO support planned on STM32F4 platform?
Now (if I right undenstend) SDIO works only on STM32F1.

tpruvot, 2019-08-28:
you may find a sample class here : https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F4/libraries/SDIO

but some F4 boards are using different frameworks, and doesnt help to implement stuff

AnHardt, 2019-08-28:
https://github.com/stm32duino/STM32SD claims to be a compete replacement for https://www.arduino.cc/en/Reference/SD working only with SDIO on the official [ST-Core](https://github.com/stm32duino/Arduino_Core_STM32).

Sadly Marlin SD-routines have been splittet and extended from its source a long time ago. So its not that easy to integrate.

tpruvot, 2019-08-29:
@AnHardt, the fat/fs is unrelated to sdio, sdio is just a kind of spi (lower level so). We use the shared SdFat/Sdcard classes on the F1 too.

AnHardt, 2019-08-29:
Have a look on how the file-system accesses the SD-Card. It works with a SDIO-connected card-slot. So the trick must be in there.

ghent360, 2019-08-31:
This code works for smt32duino & marlin 2.0:
https://github.com/ghent360/Marlin/commit/b643f285d6c2ffb9dd0c1f4f12cf46ddf5b4155f

It is small change from https://github.com/stm32duino/STM32SD

xC0000005, 2019-09-01:
I have preliminary SD support running on the lerdge boards (though it still hits errors at times) - this is using SD's SDIO block read functions.
https://github.com/xC0000005/Marlin/blob/LerdgeXAndK/Marlin/src/HAL/HAL_STM32/HAL_sdio_STM32.cpp

tpruvot, 2019-09-01:
did you try to reduce a bit the sdio clock ? can be tuned at the Hz on the F1.. should be something similar on the F4

ASPtr, 2019-09-02:
Sorry for delay. Thanks for answers.
I made a custom motherboard on STM32F407VET6 (https://github.com/jmz52/Hardware). It works fine, there is only instability in the work with the SD card. I thought SDIO would help fix this.
Unfortunately I am not a programmer, my "qualification" is only enough to write #define SDIO_SUPPORT in Configuration.h
But the firmware does not compile. More precisely compiled, but crashes at the linking stage.

`Linking .pio\build\black_stm32f407ve\firmware.elf
.pio/build/black_stm32f407ve/src/src/sd/SdBaseFile.cpp.o: In function `SdBaseFile::addDirCluster()':
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
.pio/build/black_stm32f407ve/src/src/sd/SdBaseFile.cpp.o: In function `SdBaseFile::read(void*, unsigned short)':
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:35: undefined reference to `SDIO_ReadBlock(unsigned long, unsigned char*)'
.pio/build/black_stm32f407ve/src/src/sd/SdBaseFile.cpp.o: In function `SdBaseFile::write(void const*, unsigned short)':
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
.pio/build/black_stm32f407ve/src/src/sd/SdVolume.cpp.o: In function `SdVolume::cacheFlush()':
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
.pio/build/black_stm32f407ve/src/src/sd/SdVolume.cpp.o: In function `SdVolume::cacheRawBlock(unsigned long, bool)':
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:35: undefined reference to `SDIO_ReadBlock(unsigned long, unsigned char*)'
.pio/build/black_stm32f407ve/src/src/sd/cardreader.cpp.o: In function `CardReader::initsd()':
c:\Temp\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:34: undefined reference to `SDIO_Init()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\black_stm32f407ve\firmware.elf] Error 1`

thinkyhead, 2019-09-04:
See `HAL_STM32F1/sdio.*` files for a starting-point on adding the same to `HAL_STM32_F4_F7`.

thinkyhead, 2019-09-04:
> I have preliminary SD support running on the lerdge boards

That looks promising. Hopefully we'll be able to put a general SDIO interface declaration into the `shared/` folder and then just have the `.cpp` implementations in the individual HALs.

thinkyhead, 2019-09-04:
> This code works for smt32duino & marlin 2.0:

That also looks promising. I'll make a PR.

tpruvot, 2019-09-04:
from what i seen in their "new/common" framework, they merged SDIO in a SDMMC unit, but they kept some SDIO prefixed constants

thinkyhead, 2019-09-05:
I've glommed together a number of changes from @ghent360 and elsewhere. I don't expect it all to play well together — it's just conglomeration of changes to be discussed, and the obvious keepers can be merged.

AnHardt, 2020-07-01:
Ping.

sjasonsmith, 2020-07-01:
We should hopefully on the latest stm32duino framework soon. I just have to clean up branch prior to posting that PR. That would provide the basis for whatever is needed here for HAL/STM32.

xC0000005, 2020-07-01:
I think you’ll find SDIO works with the TFT PR.

sjasonsmith, 2020-07-01:
@xC0000005 to be honest I’m not even sure what  the SDIO feature is. Is it simply to access an SD card to print, or is it something else?

xC0000005, 2020-07-01:
The SDIO peripheral can read from an SD card independent of the MCU, and DMA the data to a buffer. It’s solid hardware optimization for reading blocks of data, and the way data *should* be read when possible.

Bob-the-Kuhn, 2020-07-01:
I believe this issue can be closed.

**Sd2Card_sdio_stm32duino.cpp** implements SDIO for stm32duino boards.  Defining **SDIO_SUPPORT** in the **pins_YOUR_BOARD.h** file enables SDIO mode.

sjasonsmith, 2020-07-01:
@Bob-the-Kuhn do you know if there is anything preventing it from being enabled for common STM32F4 boards such as the SKR Pro, FYSETC S6, and Rumba32? I only see it in some less common boards like Lerdge, dev boards, that the FLYF.

I'm intending to post an update for the STM32 framework tonight. I suggest we leave this open so I can try enabling it for the boards I test with.

What would I need to do to sanity-check it? Just enable it and verify that I can still print a file from the SD card?

Bob-the-Kuhn, 2020-07-01:
None of those boards have an onboard SD card that is connected to the pins connected to the SDIO port/pins on the CPU.

sjasonsmith, 2020-07-02:
> None of those boards have an onboard SD card that is connected to the pins connected to the SDIO port/pins on the CPU.

I guess that was easy then! Yeah, I guess there is no sense to keep it open then.

It is extremely frustrating to see hardware design decisions which fly in the face of using the hardware properly. Another example is when designers spread their fan pins across as many timers as possible, either consuming all timers for PWM or forcing the use of SOFT_PWM.

@AnHardt based on the last several comments I am going to ahead and close this. You are of course free to re-open if you think there is something that was missed.

closed 2020-07-02 by sjasonsmith