MarlinFirmware/Marlin #15080: Marlin2.0-MKS-Robin2 compile error

cjsoong, 2019-08-28:
I downloaded the marlin 2.0 source code for Robin2 from MKS.
Compiling with Atom or Vscode will result in the following error.
How can I solve it?
https://github.com/makerbase-mks/MKS-Robin

[image]

cjsoong, 2019-08-28:
I solved the problem of compile error, modify platformio.ini
[env:mks_robin2]
Platform = ststm32@5.4.3
Framework = arduino
Board = MKS_Robin2
Extra_scripts = pre:buildroot/share/PlatformIO/scripts/mks_robin2.py
  Buildroot/share/PlatformIO/scripts/mks_encrypt.py
Build_flags = ${common.build_flags}
   -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"MKS_Robin2\"
   -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0xC000
Lib_deps = ${common.lib_deps}
Lib_ignore = Adafruit NeoPixel, c1921b4, TMC26XStepper, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
Src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32> +<src/HAL/HAL_STM32_F4_F7> -<src/HAL/HAL_STM32_F4_F7/*> +<src/HAL/HAL_STM32_F4_F7/STM32F4>
Monitor_speed = 250000

closed 2019-08-28 by cjsoong

github051, 2020-05-14:
Are you able to compile the marlin firmware for Robin2