MarlinFirmware/Marlin #15084: With manual mesh bed levelling, the M502 command reports G29 commands formatted for Marlin 1.x

LMF5000, 2019-08-28:
### Description

The reply for the mesh bed levelling section after sending M503 is a list of the form:
G29 S3 X2 Y1 Z-0.10000

Since Marlin 2.0 the correct syntax is I and J instead of X and Y. 

The index has to shift by -1 as well, so actually the correct format for the above should be G29 S3 I1 J0 Z-0.10000

Sending the above I1 J0 command to the printer has the desired effect of changing the Z-height of the point that the printer reports as X2 Y1.

This should be edited so that the printer reports its mesh with I J coordinates.

### Steps to Reproduce

1. Have manual mesh bed levelling enabled
2. Teach the printer a mesh using the manual method (sliding paper under nozzle and pressing OK at the right Z-height)
3. Connect to the printer in Simplify3D or similar
4. Query current settings with M503

**Expected behavior:** [What you expect to happen]

The commands reported back should be the commands sent to the printer to change the respective setting. For example the printer says "Steps per unit:
M92 X80.00 Y80.00 Z400.00 E98.00" because sending for example M92 X81 would change the setting for this line.

For mesh bed levelling, the correct command is of the form:
"G29 S3 I1 J0 Z-0.10000"

**Actual behavior:** [What actually happens]

For the above, the printer currently outputs this:
"G29 S3 X2 Y1 Z-0.10000"

#### Additional Information
In the M503, there is also no report of the Z-height (which I have currently set to 0.175mm because the nozzle scrapes the glass at Z=0)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3552420/Configuration.zip)

boelle, 2019-08-29:
if this is a bug for marlin 1.1.x then please know that its not updated anymore

use 2.0.x instead

LMF5000, 2019-08-29:
It's a bug for 2.0.x. If it was marlin 1.1.x it wouldn't be a bug ;)

boelle, 2019-08-29:
yep me that was blind, i corrected the title to reflect its a bug

boelle, 2019-09-24:
@LMF5000 is this still a problem?

LMF5000, 2019-09-24:
I've just updated to the latest marlin today. I will re-level the bed, save the eeprom data and read it back tomorrow. Will report my findings in about 24 hours.

LMF5000, 2019-09-25:
Bug still exists.

Here's what I get for M503:
Mesh Bed Leveling:
M420 S0 Z7.00
G29 S3 X1 Y1 Z0.37500
G29 S3 X2 Y1 Z0.22500
G29 S3 X3 Y1 Z0.25000
G29 S3 X4 Y1 Z0.45000
G29 S3 X1 Y2 Z0.32500
G29 S3 X2 Y2 Z0.20000
G29 S3 X3 Y2 Z0.17500
G29 S3 X4 Y2 Z0.30000
G29 S3 X1 Y3 Z0.32500
G29 S3 X2 Y3 Z0.17500
G29 S3 X3 Y3 Z0.20000
G29 S3 X4 Y3 Z0.42500

Now, say I wanted to edit the first point to Z=0.37512 instead of 0.37500

Based on the above you'd think the command to feed it is G29 S3 X1 Y1 Z0.37512

But that doesn't work.

What does work to change the first point is - G29 S3 I0 J0 Z0.37512

In summary, Marlin outputs G29 S3 Xn Yn Z-.--
It should output G29 S3 I(n-1) J(n-1) Z-.--

LMF5000, 2019-09-28:
There's another problem in the reporting of the acceleration settings. The output from the printer is like this:
Acceleration (units/s2): P<p500.00 R1000.00 T1000.00

But it should be this:
Acceleration (units/s2): M204 P500.00 R1000.00 T1000.00

boelle, 2019-10-21:
reading this again

> ### Steps to Reproduce
> 1. Have manual mesh bed levelling enabled
> 2. Teach the printer a mesh using the manual method (sliding paper under nozzle and pressing OK at the right Z-height)
> 3. Connect to the printer in Simplify3D or similar
> 4. Query current settings with M503

did you remember to store to eeprom before step 3 ? ie M500 ?

LMF5000, 2019-10-21:
Yep, settings were stored with M500 before power off.

boelle, 2019-11-03:
@LMF5000 have you tried out with 2.0.x with a copy from today?

LMF5000, 2019-11-03:
Nope, I'm using an SKR E3 DIP with TMC2208 over UART. Since they changed marlin to check the TMC stepper library I wasn't able to compile with the bigtreetech version of the library, and the marlin library didn't actually work to run the steppers. So I downgraded to the last marlin that let me run the bigtreetech library and haven't updated marlin since.

For storing the settings I'm using the hack someone mentioned in this thread to make the printer store eeprom on the SD card. It works and it saves cycles on the microcontroller's actual eeprom.

boelle, 2019-11-11:
@shitcreek @teemuatlut any idea how we can fix this one ??

boelle, 2019-11-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-24 by boelle