MarlinFirmware/Marlin #15086: [BUG] failed to compile if extruders =1 and #define STATUS_COMBINE_HEATERS

lokus77, 2019-08-28:
if you uncomment **#define STATUS_COMBINE_HEATERS** in Configuration_adv.h, with only one extruder, compilation fails with error:

status_screen_DOGM_cpp error:  **'STATUS_BED_TEXT_X' was not declared in this scope**.

zillarob, 2019-08-29:
I doubt most common boards would have a problem running 2 of the typical 30-40w heaters off one output.
If you are worried about it, or running higher wattage heaters, can use an ext mosfet like you often see for the bed.
Save that other heater output for something cool, like LEDs or fans =D

lokus77, 2019-08-29:
Is not a marlim problem, is the driver sure. Marlin with SKR V1.3 runs nice in uart mode.

Each brand (and version) of TMC2208 has different settings to enable uart. Its hard sometimes...

I did a guide:  https://docs.google.com/spreadsheets/d/13LyuaG1WTgv0dE62VP0Jh8NLjLPRxc8z2F-cb96CR2g/edit?usp=sharing
(I have tried myself  Fysetc tmc2208 V1.0  and biqu tmc2209 v3.0. The rest of tmc2208 uart soldering info comes from telegram channel, friends, or forums).

GMagician, 2019-08-29:
such define is not enough to raise compile error.
If you post your config it may help

Ludy87, 2019-08-29:
I can confirm
The bug is fixed with the PR https://github.com/MarlinFirmware/Marlin/pull/14981

boelle, 2019-09-24:
will close as PR #14981 fixed this

closed 2019-09-24 by boelle