MarlinFirmware/Marlin #15088: [BUG] Thermal runaway protection error even after changing PIDs, Works fine on stock firmware

Haxk20, 2019-08-29:
I have tevo tarantula PRO and the stock firmware works just fine with its PID. Upgraded to newest marlin 2.0 branch and updated all the pids because the stock ones were giving me runaway protection error on bed. But the same happens with the new PIDs and i basically cannot print because when i set it to 50C it reaches 41C and stays there. But weirdly when i set 60C it reaches 51C just fine and then errors out.
[Configuration stock.zip](https://github.com/MarlinFirmware/Marlin/files/3555078/Configuration.stock.zip)
[Configuration Marlin 2.0.zip](https://github.com/MarlinFirmware/Marlin/files/3555083/Configuration.Marlin.2.0.zip)
I atached zips of both stock and marlin 2.0 configs for my printer.
Marlin 2.0 configs already includes tuned PIDs.

boelle, 2019-08-29:
could it be that the bed is simply just to slow?

try and use a stop watch and test how fast it reach's 40, 50, 60.... Does it take longer to reach 60 than 40? if so i would vote on the bed is just to weak

Haxk20, 2019-08-29:
The bed doesnt have this issue on marlin 1 which is stock firmware and has thermal runaway protection enabled too so the bed is OK. 
The issue here is that the bed stops always 9C below the target and then throws out error. If i set target to 70C it would get to 61C and after minute or so will error. If i set it to 60C it would reach 51C and so on.

boelle, 2019-08-29:
is #define THERMAL_PROTECTION_BED_PERIOD also set to the same values?

thisiskeithb, 2019-08-29:
In `Configuration stock.zip`, `PIDTEMPBED` is commented out ([Tevo's Tarantula Pro repo](https://github.com/tevo3d/Tarantula-pro) has `PIDTEMPBED` disabled as well), so it's using bang-bang to heat the bed. I can't open your `Configuration Marlin 2.0.zip`, but the Tarantula Pro config in `bugfix-2.0.x` [has it enabled](https://github.com/MarlinFirmware/Marlin/blob/38983fdfa9773faf7006ef392110aca01ad006d5/config/examples/Tevo/Tarantula%20Pro/Configuration.h#L515) with default (untuned) values. 

If you comment out `PIDTEMPBED`, does the bed heat as it should?

Alternatively, you can find your own bed PID values by running a PID tune. With the bed at room temperature (this is important) run `M303 C6 E-1 S60`.

So we have a couple options:

1. Update the Tarantula Pro config using your PID values when you run a PID tune on the bed or
2. Disable `PIDTEMPBED` on the Tarantula Pro config and let it use bang-bang if that works fine

Haxk20, 2019-08-29:
Im in the middle of big print using stock firmware so i will try once the print is done.

Haxk20, 2019-08-29:
@thisiskeithb But as i said i updated the PIDs in the new configuration.h and uploaded. It does the same thing.

thisiskeithb, 2019-08-29:
I'm leaning toward disabling `PIDTEMPBED` in the config.

boelle, 2019-08-29:
same her, just did not think of disabling pid for the bed

what could be the reason for them doing that?

thisiskeithb, 2019-08-29:
I put in a PR (https://github.com/MarlinFirmware/Marlin/pull/15089) to disable `PIDTEMPBED` since that is what stock uses.

Users can always enable it and run a PID tune to get values for their machine if so desired.

Haxk20, 2019-08-29:
Confirming. Issue is fixed.

closed 2019-08-29 by Haxk20