MarlinFirmware/Marlin #15090: [BUG] Compiler warnings

LineF, 2019-08-29:
In bugfix-2.0.x branch there is a compiler warning issue since a longer time (i think somewhere in 2018):

In many compiled programs I get following warning (Arduino IDE 1.8.7):

===============
In file included from C:\Users\Martin\AppData\Local\Temp\arduino_build_976427\sketch\src\gcode\temperature\M140_M190.cpp:30:0:

c:\users\martin\appdata\local\temp\arduino_build_976427\sketch\src\lcd\ultralcd.h:343:29: warning: extra qualification 'MarlinUI::' on member 'expire_status_ms' [-fpermissive]

             static millis_t MarlinUI::expire_status_ms; // = 0
===============

Maybe someone can find the issue...

Thanks,
Martin

RudyBenner, 2019-08-29:
Try Arduino IDE 1.8.9

boelle, 2019-08-29:
or 1.9 or vs code with platform.io extension

LineF, 2019-08-29:
no, IDE 1.9.0 beta doesn't help - the warning is still present.
What do you mean with "vs code with platform.io extension" ?

boelle, 2019-08-29:
https://code.visualstudio.com/

LineF, 2019-08-29:
@ludy87: the fix solved the problem!
Thx a lot!

Ludy87, 2019-08-29:
@boelle i can confirm

boelle, 2019-08-29:
@Ludy87 oki, we can close this one when the PR is merged

now you just need to make fix's for the other bugs :-D

Ludy87, 2019-08-29:
@boelle Which mistakes are that?

boelle, 2019-08-29:
https://github.com/MarlinFirmware/Marlin/issues/15086
https://github.com/MarlinFirmware/Marlin/issues/15084
https://github.com/MarlinFirmware/Marlin/issues/15083

just examples.... :-D

Ludy87, 2019-08-29:
For #15086 I already wrote a PR ;-P

Ludy87, 2019-08-29:
@boelle
then it can be done here, Scott has brought the PR together

boelle, 2019-08-30:
@Ludy87 it was meant like a joke :-D we have a lut of bugs that needs a smart head or maybe 10 smart heads

but of course an issue list without bugs is never going to happen, when new things are added the risk of  bugs is there

boelle, 2019-08-30:
pr is merged, closing

closed 2019-08-30 by boelle