MarlinFirmware/Marlin #15095: [BUG] (bugfix-2.0.x Teensy 3.6 + FYSETC_MINI_12864_2_1 hardware SPI compile error)

Arotus-3D, 2019-08-29:
### Description

I am using Teensy 3.6 with FYSETC Mini12864 V2.1
When i try to complie it in hardware SPI mode i get the compiler error.

### Steps to Reproduce

1. Install
arduino 1.8.9
teensyduino 1.47
U8glib 1.19.1 and TMCStepper 0.4.6 from Library Manager
(use servo and neopixel lib from teensyduino)
2. copy Configuration.zip to the project
3. copy pins_TEENSY35_36.h to Marlin\src\pins\teensy3
4. complie

**Expected behavior:**

Good compile and LCD working on hardware SPI.

**Actual behavior:**
I get this error

sketch\src\lcd\dogm\u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_hw_spi+0x8): undefined reference to `u8g_com_arduino_hw_spi_fn'

collect2.exe: error: ld returned 1 exit status

#### Additional Information

It compiles when i do `#define FORCE_SOFT_SPI` in pins_TEENSY35_36.h. In this mode the LCD is working.
Aditional i use three TMC2130(software spi at the moment) and SDCARD.
All devices are on the same SPI bus with separate CS pins

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3555853/Configuration.zip)

shitcreek, 2019-09-03:
This is an external library issue which isn't in our control. You could modify the  u8g_com_arduino_hw_spi.c specific to your board if you want to get it working. I found this discussion which might help although it pertains to Teensy 3.0 - 3.2: https://forum.pjrc.com/threads/23445-Teensy-3-0-and-u8glib/page4

boelle, 2019-09-24:
will close this as it out of marlin firmware control

closed 2019-09-24 by boelle