MarlinFirmware/Marlin #15096: [BUG] MKS Gen L TCM2208 UART connection issue

porlock, 2019-08-29:
### Description
I have a issue with connection over UARTto tcm2208 drivers , sometimes g-command m122 shows zeros for driver register address, it happens randomly for random axis, when I run m122 again is fine, it happens real randomly so I don't think this is a wiring fault. My board is MKS Gen L and I run marlin 1.1 from bugfix branch. What is real strange that i noticed too, is that some times when i run m122, probe of my bltouch sensor needle is deployed and goes up, like when probing.  I check wires many times, take off the plastic shell and use calipers to make it tighter, now they are holding real good. It is not possible that all cables are bad, and it fail in random order. I have 2 motors for Z axis that is why i utilize two extra pins form aux port for software UART

### Steps to Reproduce

1. Compile marlin from buxfix branch with my config files
2. Install marlin on mks gen l with tcm2208 and bltouch
3. Connect tcm2208  to the board with software uart pins.
4. Power cycle printer
5. Run m122  10 times in the row
6. Check driver register address for all zeros adress
7. Check behavior of bltouch needle .

#### Additional Information
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3557525/configs.zip)

shitcreek, 2019-09-02:
Marlin 1.x is no longer being worked on. Please try Marlin 2.0 and let us know if the problem still exists. Thank you.

porlock, 2019-09-11:
On Marlin 2.0 it happens some time to time but is its 80% better.

closed 2019-09-11 by porlock