MarlinFirmware/Marlin #15097: [FR] Stop print when communication UART/SPI with TMC steper drivers fails

porlock, 2019-08-29:
It will be good to add an option for automatically stop print when communication with step driver over UART or SPI fails and settings for motor current, working mode etc. can not be applied. 
BTW when this settings are apply ? , when printer is turned  or maye when print start ? is there one time action or there are retries ?

ManuelMcLure, 2019-08-29:
As far as I can tell, the settings are sent to the drivers in the following cases:
* After EEPROM settings are read from disk or initialized either with ``M501``, ``M502``, or on bootup.
* When ``PSU_CONTROL`` is enabled and the power supply is turned on either explicitly through ``M80`` or implicitly via ``AUTO_POWER_CONTROL``.
* When a gcode that explicitly sets a driver attribute is used (such as ``M569``).

The first two will attempt to completely initialize or re-initialize the drivers. The latter will only set the relevant setting.

porlock, 2019-08-29:
Unfortunately there is no communication verification :-( so we don't know is setting are successfully set on step driver or not. What do you think about new gcode command that will perform that verification and stop the print if its fial ?