MarlinFirmware/Marlin #15102: Unable to compile Marlin 2.0.X with Arduino IDE 1.8 or 1.9

anjoze, 2019-08-30:
Hi, I'm trying to compile Marlin 2.0.x with Arduino 1.8.9-linux64, Arduino 1.9.0-beta-linux64 and Arduino 1.8.9-windows, and I get always the same error:
 
_sketch/src/module/stepper/TMC26X.cpp:28:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
 #include "../inc/MarlinConfig.h"
compilation terminated.
exit status 1
Error compiling to board Arduino/Genuino Mega or Mega 2560._

sl1pkn07, 2019-08-30:
try platformio

greetings

DkOuvE, 2019-08-30:
Yeah I'm getting this too. For that error, it's just because it's supposed to be ../../MarlinConfig.h. I think there's one more file that needs that change. It should tell you when you compile.

But then I get the error where __TMC_CLASS requires 4 arguments, but only 3 are given.

sl1pkn07, 2019-08-30:
yep, seems platformio also affected

~~~
Compiling .pio/build/LPC1768/src/src/module/stepper/trinamic.cpp.o
Marlin/src/module/stepper/TMC26X.cpp:28:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
 #include "../inc/MarlinConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/LPC1768/src/src/module/stepper_indirection.cpp.o
*** [.pio/build/LPC1768/src/src/module/stepper/TMC26X.cpp.o] Error 1
In file included from Marlin/src/module/stepper/indirection.h:40,
                 from Marlin/src/module/stepper/indirection.cpp:34:
Marlin/src/module/stepper/trinamic.h:29:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
 #include "../inc/MarlinConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/LPC1768/src/src/module/stepper/indirection.cpp.o] Error 1
In file included from Marlin/src/module/stepper/trinamic.cpp:32:
Marlin/src/module/stepper/trinamic.h:29:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
 #include "../inc/MarlinConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/LPC1768/src/src/module/stepper/trinamic.cpp.o] Error 1
~~~

Ludy87, 2019-08-30:
change to `#include "../../inc/MarlinConfig.h"`

Bergerac56, 2019-08-30:
If not already found, it seems that the #include "../inc/MarlinConfig.h" of the TMC26x.cpp is misspelled

Replace it by ../../inc/MarlinConfig.h (../ missing)

Bergerac56, 2019-08-30:
@Ludy87 Sorry. Sent my 2 cents before seeing your post ;)

sl1pkn07, 2019-08-30:
after
~~~patch
diff --git a/Marlin/src/module/stepper/TMC26X.cpp b/Marlin/src/module/stepper/TMC26X.cpp
index 30b01ab4f..fb7007873 100644
--- a/Marlin/src/module/stepper/TMC26X.cpp
+++ b/Marlin/src/module/stepper/TMC26X.cpp
@@ -25,7 +25,7 @@
  * Stepper driver indirection for TMC26X drivers
  */
 
-#include "../inc/MarlinConfig.h"
+#include "../../inc/MarlinConfig.h"
 
 //
 // TMC26X Driver objects and inits
diff --git a/Marlin/src/module/stepper/trinamic.h b/Marlin/src/module/stepper/trinamic.h
index 5f0ef6d71..181f195f6 100644
--- a/Marlin/src/module/stepper/trinamic.h
+++ b/Marlin/src/module/stepper/trinamic.h
@@ -26,7 +26,7 @@
  * Stepper driver indirection for Trinamic drivers
  */
 
-#include "../inc/MarlinConfig.h"
+#include "../../inc/MarlinConfig.h"
 
 #include <TMCStepper.h>
 #if TMCSTEPPER_VERSION < 0x000405
~~~

'''
In file included from Marlin/src/module/stepper/indirection.h:40,
                 from Marlin/src/module/stepper/indirection.cpp:34:
Marlin/src/module/stepper/trinamic.h:66:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS(X, X) stepperX;
 ^ ~~~~
Marlin/src/module/stepper/trinamic.h:78:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS(Y, Y) stepperY;
 ^ ~~~~
Marlin/src/module/stepper/trinamic.h:90:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS(Z, Z) stepperZ;
 ^ ~~~~
Marlin/src/module/stepper/trinamic.h:150:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS_E(0) stepperE0;
 ^ ~~~~
In file included from Marlin/src/module/stepper/indirection.cpp:34:
Marlin/src/module/stepper/indirection.h:63:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
     extern TMC_CLASS(X, X) stepperX;
 ^   ~~
compilation terminated due to -fmax-errors=5.
In file included from Marlin/src/module/stepper/trinamic.cpp:32:
Marlin/src/module/stepper/trinamic.h:66:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS(X, X) stepperX;
 ^ ~~~~
Marlin/src/module/stepper/trinamic.h:78:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS(Y, Y) stepperY;
 ^ ~~~~
Marlin/src/module/stepper/trinamic.h:90:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS(Z, Z) stepperZ;
 ^ ~~~~
Marlin/src/module/stepper/trinamic.h:150:1: error: macro "__TMC_CLASS" requires 4 arguments, but only 3 given
   extern TMC_CLASS_E(0) stepperE0;
 ^ ~~~~

'''

Ludy87, 2019-08-30:
next bug level :-D

Ludy87, 2019-08-30:
I think that should look at @thinkyhead and @teemuatlut

At the PR https://github.com/MarlinFirmware/Marlin/commit/4758db6f9c082c6c73ed462b19e996f1ed4ece04 something went wrong.

Stay calm and wait

leseaw, 2019-08-30:
What is this tip for deleting the folder? If he now has TMCs and then?

The better tip is at the top, replace line.

klcjr89, 2019-08-31:
This fails to compile in VSCode as well. Please fix this issue asap!

sl1pkn07, 2019-09-01:
seems fixed (seems this commit is forced pushed https://github.com/MarlinFirmware/Marlin/commit/87b16ed5f65bb8511d648c47a3d48fadf1f2dee7). failed when rebase in my tree

leseaw, 2019-09-03:
looks okay for me now.
i think we can close this here.

closed 2019-09-03 by shitcreek

leytanelectronics2, 2020-05-31:
@Ludy87  > change to `#include "../../inc/MarlinConfig.h"`

does this apply to all instances of this line in the whole script? or is it only in one folder?