MarlinFirmware/Marlin #15103: [FR] Thermal runaway protection shouldn't be linear when approaching maxtemp

LinoBarreca, 2019-08-30:
### Description

When working close to the temperature limits of the hardware the thermal runaway protection shouldn't be a linear function but rather logarithmic.
There should be a parameter above which the thermal protection isn't checked linearly but logarithmically.

### Steps to Reproduce

1. disable thermal runaway protection for your bed.
2. set a very high maxtemp for the bed and flash the firmware.
3. set the bed to a ridiculously high temperature (let's say 200C°) for 15 minutes. take note of the real max temp of the bed after 15 minutes. Mine is 138°C.
4. set in firmware maxtemp = 140.
5. re enable thermal runaway protection with reasonable parameters.
6. upload firmware.
7. try to heat the bed to 135°C.
8. the bed temp will raise a pair of degrees per minute until around 110° then the temperature increase will slowly decelerate (this is expected and natural, it's like a car reaching its max speed)
9. around 120° you will have the message "heating failed" because the temperature didn't rise as fast as merlin expected (but still according to physical laws)

#### Additional Information

none.

AnHardt, 2019-08-30:
Nice idea but nonsense.
When the maximum temperature a heater can reach at full power is below the temperature it can stand for a long time, `THERMAL_PROTECTION` is not needed (and does not work). 
**If it's for a bed - Deactivate it!**

`MAXTEMP` is the maximum temperature a heater shall heat to without any part degrading (melting, burning, smelling, ...). Together with MINTEMP we can detect if thermistor is electrically shorted or disconnected.
*Self-limiting-temperature* is the temperature a heater can reach when full powered for a long time.

`THERMAL_PROTECTION` deals not with the electrical, but with the thermal connection between heater and sensor. When the sensor comes off the heater, it will measure a much lower temperature than the heater has. That, at the worst case, will full power the heater.

If the heater is self-limiting below `MAXTEMP` nothing bad can happen. And only in that case the temperature raise per time is going to zero.

For the extruder heaters there is a failure where the heater cartridge falls out of the heater-block.
**Never deactivate `THERMAL_PROTECTION` for a extruder heater.** But usually extruder heater systems have enough power to limit their temperature by self destruction - so above `MAXTEMP`.

InsanityAutomation, 2019-08-30:
> Nice idea but nonsense.
> When the maximum temperature a heater can reach at full power is below the temperature it can stand for a long time, `THERMAL_PROTECTION` is not needed (and does not work).
> **If it's for a bed - Deactivate it!**
> 
 I absolutely disagree with that statement. In a timely fashion, I just worked with a user today who saw the bed shutting off due to thermal runaway where it normally holds fine. Found the connector for it was hot, a solder joint broke and was arcing to the housing. Thermal runaway shut it down before it got hot enough to potentially burn the connector.

LinoBarreca, 2019-08-30:
Well... no. Sorry to disagree. Let me explain why.
If you intend the thermal protection as a way to not burn your house you are completely right. And I agree with you. Since the bed can't go above a certain limit (because it dissipates heat) it shouldn't make sense to have such a check...

For what I see, however, the thermal protection isn't just that but rather a way to tell the user "hey, dude there's something wrong with the heating of this part" (even if it isn't fire-dangerous).
It could be a lose contact on the heater cables, it could be that you are keeping the printer in a too cold environment, it could be a temperature sensor broken and so on.
Considering the thermal protection feature this way, keeping it on for a component which limites it's temperature "naturally" (like the bed) actually makes sense..... as long as it works properly according to law physics.
Just consider a CLOSED building chamber. If something goes wrong and you have that disabled the whole building chamber will get to the maximum bed temperature (almost like an oven) and that is bad (for examples for motors and cables)

LinoBarreca, 2019-08-30:
@AnHardt see?
@InsanityAutomation wrote exactly what I meant while I was writing my comment.
Thermal protection is more than "I will not heat that up to its melting point" :)
It's a way to check/verify that the temperatures asked are met and correct.

AnHardt, 2019-08-31:
Gentlemen.
Please reread my definitions.
If you try to reach a temperature above the temperature the heater can reach, because at this temperature the losses are bigger than the heaters power, a algorithm based on a rise of temperature in time, can not work because the rise will be zero before the target temperature is reached. Regardless of the scale.

What i did not talked about - the the self-limiting-temperature depends on the temperature of the environment. Today under the roof i have 40°C and may bed may be able to reach 150°C, while in the winter with 10°C it may reach 120°C. Newtons law of cooling says - it's a fixed number above the environments temperature.

Newtons law of cooling may be the base of a next generation heater algorithm and failure detection bu it's not yet ready for public. An maybe to calculation intensive or memory hungry for the 8-bitters.

AnHardt, 2019-08-31:
> When the maximum temperature a heater can reach at full power is below the temperature it can stand for a long time, THERMAL_PROTECTION is not needed (and does not work).
If it's for a bed - Deactivate it!

The implizite alternative is: "Set `MAXTEMP` **well below** the *self-limit-temperature*."
Then you can't come into a situation the algorithm can't handle.

boelle, 2019-08-31:
so is this a bug? feature request or user misunderstanding?

LinoBarreca, 2019-09-05:
> so is this a bug? feature request or user misunderstanding?

It's a feature request for something that has been implemented in a too simplistic way which causes problems in some circumstances. :)

AnHardt, 2019-09-05:
I guess it's ignoring physics in point 5 (how to repeat). *Reasonable parameters* is the keyword. Close to the *Self-limiting-temperature* there are no *reasonable parameters*, regardless of the skale. At the  *Self-limiting-temperature* the rise is zero (and above negative). The *reasonable parameters* have to match the worst case, what they obviously did not (and can not at and above the *Self-limiting-temperature*).
So (for me) user misunderstanding of the concept.

-----------
The *increase values* absolute lower limit is the noise level of the temperature measurement. Below that you can't reliably measure temperature differences. Set about the double amount.
The worst case can't be taken from looking at the continuous heating curve. That would miss the additional delays when switching from cooling to heating. The best procedure, i know, is: 
Heat to maximum temperature you want to print with (on a cool day, with fans on - worst case). Switch off the heater until you see the temperature falling to *increase value* below max. Switch on again and measure the time until heater is at temperature again. Set that time plus a safety margin. 
Else you will get false positives sooner or later.
Repeat the procedure when changing your opinion about the max temperature you want to print with or the heater system is changed. (Some thermal grease, or a tiny sock do make a large difference.)

AnHardt, 2019-09-05:
Let me plot the idea for the alternate heater algorithm.
Let's begin with an experiment.
Ideally take a self limiting bed heater.
Begin with the heaters temperature at the environments temperature.
Set `#define PID_OPENLOOP 1`, to have direct access to the heater power.
Set MAXTEMP very high and switch off all protections. (So be careful and stay close to the machine.)
Let's protocol the temperatures and heater power over time. (Repetier Host Temperature Plot).

1. Switch the heater full on.
2. When the temperature does not rise anymore (or you are scared to ruin the heater) switch off the heater.
3. When the temperature does not fall anymore set heater power to 10%.
4. When the temperature does not rise anymore set heater power to 20%.
5. When the temperature does not rise anymore set heater power to 30%.
and so on to 100%.
12. Now again at 100% and max temperature, switch on a fan blowing over the surface of the heater. Observe the temperature falling to a new stable value below the former max.
13: Switch off the heater and restore safe settings.

At first let's have a look on the plot of phase 2. After a few seconds we get a smooth curve. At the beginning falling fast, then more and more slowly falling arriving room temperature. What we see is the temperature loss over time. That is about proportional the energy loss over time. For every temperature above room temperature and below the maximum temperature we can find a value in that graph - how many °C is the temperature dropping in the next second. The curve can be calculated with [Newton's law of cooling](https://en.wikipedia.org/wiki/Newton%27s_law_of_cooling).
[image]
The temperatur `T` at the time `t` T(t) is equal to the room temperature `Tenv` plus the difference of the start (max) temperature `T(0)` and the room temperature `Tenv` times e (a mathematical constant)
to the power of the product of minus an other konstant `r` and the time `t` since the cooling started.
In `r` we have the properties of the heater system, like surface of the heater and the surrounding media and how fast that moves.

In phase 1 we see power the heater with a constant wattage. If we would not have energy losses, we'd see a straight line from the start to infinity. For every second the heater is on we'd see a constant rise in temperature.
There are losses. The losses are the same as in part 2 of the graph. So after a second of heating we get a constant rise of the temperature but also a temperature dependent decrease what we can get from phase 2. 
At some temperature (the self-limit-temperature) the losses are exactly that high as the heater power.

In the phases 3. to 11. we see about the same as in phase 1. The temperature rises slowly to that temperature where the losses are equal to the energy input. We also observe the distance between the temperatures gets smaller, in the equal 10% power steps.
What is not that obvious is that the power we put in the system is represented exactly the I therm in a PID. When the PID has swung in, is and target temperature is the same P is 0 and if that is for some time D is 0. Only I is remaining.

In phase 13. we see what happens if `r` begins to differ. A PID would after some time adjust its I therm. to get back to the former temperature.

What can we do with that insights.
From phase 2. we can determine `r`. From phase 1. and `r` we can determine the wattage. With 'r` and the wattage we can correlate wattage and target temperature.

For now let's assume we don't have disturbances.
When we want to heat from T1 to T2 we can look up the duration for heating from phase 1 and give full power for that time. After that time we can switch to a heater power looked up from phase 3 to 11. When cooling from T2 to T3 we can look up the duration for switching the power off in phase 2. There after we can again heat with a power looked up in 3 - 11.
In theory we can give or take exactly that amount of energy to reach the new temperature, without overshot.
 
If we have disturbances we could either put a PD controller on top - at least when holding temperatures, or a PID and derive `r` from the I-term, or we could recalculate the actual `r` from the temperatures and heater powers of the last few seconds. But because of the delay between applying the new power and the reaction of temperature this is not trivial. 

For fan off and fan on we will get different but similar values for `r`. When we get a problem with the heater, like separated, shorted, broken sensors or heater cartridges, `r` will differ a lot, so errors can be detected.

thinkyhead, 2020-07-03:
I'm all in favor of adding something to the future "Printer Setup" menu which will watch the heater and bed temperature similar to `M303`, but which will take a look at your typical temperature rise over time and tune the temperature watchdog within a reasonable margin of common conditions.