MarlinFirmware/Marlin #15106: [BUG] (short description)

NorxMAL, 2019-08-30:
### Description

Got error on not finding a file, and multiple definitions of 'restore_stepper_drivers()'

### Steps to My FIX

Change line 28 in src\module\stepper\TMC26X.cpp to :
#include "../../inc/MarlinConfig.h"

This is not a fix, just a bandaid:
src\module\stepper_indirection.cpp
and
src\module\stepper\indirection.cpp

Yeah, the latter file has just has 
```
  #if HAS_TRINAMIC
    restore_stepper_drivers();
  #endif
```
in restore_stepper_drivers(), which the former doesn't have. 

And in reset_stepper_drivers(), the latter (indirection.cpp) has
```
  #if HAS_TRINAMIC
    reset_trinamic_drivers();
  #endif
```
And former (stepper_indirection.cpp has
```
  #if HAS_TRINAMIC
    static constexpr bool stealthchop_by_axis[] = {
      #if ENABLED(STEALTHCHOP_XY)
        true
      #else
        false
      #endif
      ,
      #if ENABLED(STEALTHCHOP_Z)
        true
      #else
        false
      #endif
      ,
      #if ENABLED(STEALTHCHOP_E)
        true
      #else
        false
      #endif
    };
  #endif
```

I just deleted both reset_stepper_drivers() and reset_stepper_drivers() code sections in indirection.cpp (The file is essentially empty) and added 
```
  #if HAS_TRINAMIC
    restore_stepper_drivers();
  #endif
```
to stepper_indirection.cpp restore_stepper_drivers(); so it looks like this 
```
¨void restore_stepper_drivers() {
  #if AXIS_IS_TMC(X)
    stepperX.push();
  #endif
  #if AXIS_IS_TMC(X2)
    stepperX2.push();
  #endif
  #if AXIS_IS_TMC(Y)
    stepperY.push();
  #endif
  #if AXIS_IS_TMC(Y2)
    stepperY2.push();
  #endif
  #if AXIS_IS_TMC(Z)
    stepperZ.push();
  #endif
  #if AXIS_IS_TMC(Z2)
    stepperZ2.push();
  #endif
  #if AXIS_IS_TMC(Z3)
    stepperZ3.push();
  #endif
  #if AXIS_IS_TMC(E0)
    stepperE0.push();
  #endif
  #if AXIS_IS_TMC(E1)
    stepperE1.push();
  #endif
  #if AXIS_IS_TMC(E2)
    stepperE2.push();
  #endif
  #if AXIS_IS_TMC(E3)
    stepperE3.push();
  #endif
  #if AXIS_IS_TMC(E4)
    stepperE4.push();
  #endif
  #if AXIS_IS_TMC(E5)
    stepperE5.push();
  #endif
  #if HAS_TRINAMIC
    restore_stepper_drivers();
  #endif
}
```

And didn't do anything to restore_stepper_drivers() in stepper_indirection.cpp

Ludy87, 2019-08-30:
https://github.com/MarlinFirmware/Marlin/issues/15102

boelle, 2019-08-31:
please edit the title to be a bit mor meaningfull

NorxMAL, 2019-09-01:
> 
> 
> #15102

Where does it say anything about src\module\stepper_indirection.cpp and src\module\stepper\indirection.cpp?

NorxMAL, 2019-09-01:
> 
> 
> please edit the title to be a bit mor meaningfull

People find this through google anyhow :P

Ludy87, 2019-09-01:
See and try https://github.com/MarlinFirmware/Marlin/pull/15111

boelle, 2019-09-24:
@NorxMAL is this still a problem?

NorxMAL, 2019-09-24:
@boelle No, it is running fine on my Delta Ramps1.4 now :)

boelle, 2019-09-25:
oki, will close then, we can always reopen

closed 2019-09-25 by boelle