MarlinFirmware/Marlin #15108: SKR 1.3 & TMC 2208 Homing

stephenmott, 2019-08-30:
### Description

Latest pull (just now), Just trying to upgrade my drivers from DRV8825 to TMC2208's on a Cetus3D upgrade, fairly smooth so far, except bed leveling, it's very odd, you level it, it says it's failed, but the screen says it's gone 50/50 in X/Y, which isn't correct, but it's actually only gone (estimated) 5/5, so the probe is way off the bed, micrometer measuring distances moved, seems fine, so I don't think it's a steps per mm problem.. it worked fine on the DRV8825's, so I assume it's a TMC/SKR 1.3 issue, everything else seems fine?

(BTW, not related, but tried to compile SRK 1.3 & MKS_12864OLED -  nightmare, went back to a normal full graphics display, many missing defines, probably need a more sensible error if it isn't supported.. )

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3561629/Archive.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-08-30:
Try changing your microsteps to 32 (TMC2208s interpolate to 256).

stephenmott, 2019-08-31:
> Try changing your microsteps to 32 (TMC2208s interpolate to 256).
> 
> Edit: What's the default steps/mm for this printer & are the steppers 1.8° or 0.9°? `DEFAULT_AXIS_STEPS_PER_UNIT` of 1280 seems high.

Not 200% sure, there's nothing written on them. The extruder is 0.9 with a geared BTech clone.

I tried from 16 up to 128 and that works, it's just the 256 one that doesn't. It's now using

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 640, 640, 640, 7500 }`

thisiskeithb, 2019-08-31:
256 native microstepping is A LOT, even for 32-bit. See this comment for more info: https://github.com/MarlinFirmware/Marlin/issues/14298#issuecomment-502434434

thisiskeithb, 2019-08-31:
Is this the same machine? https://github.com/etet100/marlin_cetus_3d

If so, default steps are: `DEFAULT_AXIS_STEPS_PER_UNIT   { 160, 160, 160, 225 }`

stephenmott, 2019-08-31:
Yea, I'm testing now, and it's not what I'd call 'stable', missing steps, drifting..

stephenmott, 2019-08-31:
> Is this the same machine? https://github.com/etet100/marlin_cetus_3d
> 
> If so, default steps are: `DEFAULT_AXIS_STEPS_PER_UNIT { 160, 160, 160, 225 }`
> 
> Edit: Seems likely since 640 is a multiple (4x) of 160.

Yea, looks similar, but different CPU, steppers etc..

thisiskeithb, 2019-08-31:
Whatever the exact model is, you have to match your steps/mm & microstepping. Try [Prusa's RepRap calculator (Stepper Motors)](https://blog.prusaprinters.org/calculator/#steppermotors) tool to help figure out the correct combo.

stephenmott, 2019-08-31:
That's OK if you've got the numbers to plug into it, I find it simpler to just measure it and do the maths..

shitcreek, 2019-09-05:
@stephenmott - using 256 steps will needlessly burden your board when the stepper driver will interpolate to 256 steps anyway.

stolen from https://forum.duet3d.com/topic/3980/correct-microstepping-settings/13:
> No micro stepping:
> GCODE "Move X 1mm" => Duet "calculate 100 step pulses" => TMC Stepper Driver "100 step pulses" => Motor "1mm"
> x16 micro stepping:
> GCODE "Move X 1mm" => Duet "calculate 1600 micro-step pulses" => TMC Stepper Driver "1600 micro-step pulses" => Motor "1mm"
> x16 micro stepping with interpolation to x256:
> GCODE "Move X 1mm" => Duet "calculate 1600 micro-step pulses" => TMC Stepper Driver "25,600 micro-step pulses" => Motor "1mm"
> x256 micro stepping:
> GCODE "Move X 1mm" => Duet "calculate 25,600 micro-step pulses" => TMC Stepper Driver "25,600 micro-step pulses" => Motor "1mm"

stephenmott, 2019-09-05:
With interpolation on, when I set it to 16, there was a really annoying high pitch stepper sound.. I've currently got it set to 32 with interpolation for XYZ and I've gone back to DRV8825 for E0 as I couldn't get the TMC's to not miss steps extruding (stealth chop or spread cycle).

shitcreek, 2019-09-05:
TMC2208 has stealthChop2 which specifically addresses the misstepping issue.
Has you adjusted your current / vref? This is what I am using with my 2208's:
X , Y = 600, E = 900, Z = 1000.
Make sure you do M502 and M500 after flashing.

stephenmott, 2019-09-05:
Is that something you have to enable, I can't see a setting for it? My X/Y/Z are all 160 at the moment, I'm not doing the M502/500, but I am resetting/storing it via the menu.

shitcreek, 2019-09-06:
I think you may be confusing steps with current (voltage).

In the section `#if HAS_TRINAMIC` of the `configuration_adv.h` file,
```
  #if AXIS_IS_TMC(X)
    #define X_CURRENT     800  // (mA) RMS current. Multiply by 1.414 for peak current.
```
You can use `M92` to override them or via the menu. Use `M503` to check what they are.

stephenmott, 2019-09-06:
I don't _think_ so, steps are:

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 160, 160, 160, 640 }`

Extruder was:

```
#if AXIS_IS_TMC(E1)
    #define E1_CURRENT   1200
    #define E1_MICROSTEPS 16
    #define E1_RSENSE    0.11
  #endif
```

Stealthchop (1??) is turned on via

```
#define STEALTHCHOP_XY
#define STEALTHCHOP_Z
#define STEALTHCHOP_E
```
I can't see anything about stealthchop 2..

shitcreek, 2019-09-06:
StealthChop is a feature of the stepper driver: https://www.trinamic.com/technology/adv-technologies/stealthchop/

Shouldn't you be using E0?

closed 2019-09-06 by stephenmott