MarlinFirmware/Marlin #15110: [BUG] (SKR 1.3 Dual Z Stepper) Z2 does not work

CarlosLB92, 2019-08-31:
Hello, I have a P802MA printer, which I have updated time ago with a SKR 1.3, when I first mounted this board it was with the DRv8825 using the E1 as Z2, but recently I acquired the TMC 2130, but when trying to install them I have been able to use the dual Z stepper driver because the Z2 does not move, I have tried to change the TMC2130 axis and it works correctly. I haven't activated the less homing sensor yet.

[Marlin skr1.3 p802ma tmc 2130 dual Z.zip](https://github.com/MarlinFirmware/Marlin/files/3561963/Marlin.skr1.3.p802ma.tmc.2130.dual.Z.zip)

Patag, 2019-08-31:
Hi,
Try to comment `//#define E1_DRIVER_TYPE TMC2130`

jared248371, 2019-09-01:
I'm having this same exact issue except with the TMC2208.
I've tried configuring it with both #define Z2_DRIVER_TYPE TMC2208 and #define E1_DRIVER_TYPE TMC2208.
Configuration_adv contains #define Z_DUAL_STEPPER_DRIVERS

From another post on here's suggestion, I even tried updating pins.h to include Z2 instead: 
//#define E1_SERIAL_TX_PIN P1_04
//#define E1_SERIAL_RX_PIN P1_01 

#define Z2_SERIAL_TX_PIN P1_04
#define Z2_SERIAL_RX_PIN P1_01
No matter what, my second z motor will not turn. Let me know if you have any breakthroughs on this.

GizmoTheGreen, 2019-09-01:
> Hello, I have a P802MA printer, which I have updated time ago with a SKR 1.3, when I first mounted this board it was with the DRv8825 using the E1 as Z2, but recently I acquired the TMC 2130, but [...]

whats your output of m122?

Bob-the-Kuhn, 2019-09-01:
I just started up a TMC5160 system with dual Z steppers.  Initially they didn't turn because the current was too low because my SPI was not cabled up properly.  Does M122 report that coms is OK for all the TMC2130s?

shitcreek, 2019-09-03:
Do not enable `E1_DRIVER_TYPE`. You only need `Z2_DRIVER_TYPE` and `Z_DUAL_STEPPER_DRIVERS`.

shitcreek, 2019-09-04:
Closing this issue because we have this very board running dual z motors without any issue on one of our printers. I will gladly reopen it if anyone begs to differ.

closed 2019-09-04 by shitcreek

gnattycole, 2020-01-01:
Folks, I'm having the same issues as the original post and I'm wracking my brains how to get it fixed. I'm using TMC2208 on all axis including extruder. I have 2nd Z axis motor plugged into E1 port on the SKR 1.3. In config_ADV I have set the Z_DUAL_STEPPER_DRIVERS to enable In config .h, I have tried setting the Z2 to TMC2208 and enabling. It compiles ok but once uploaded to the board, I get a TMC comunication error message on the LCD. If I enable the E1 and assign the TMC2208 to it, it compiles fine and the comms error goes away, however the two Z axis motors turn at different rates, causing the bed to skew over. I tried enabling DISTINCT_E_FACTORS and added a 5th term to the DEFAULT_AXIS_STEPS_PER_UNIT _"80, 80, 400, 90, 400"_ but I then get a compile error : Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:2265:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT requires X, Y, Z and E elements.
Ive tried with various different combinations of commas, brackets etc but not sure if it's just a syntax issue or whether I'm missing something else. The machine is a Tronxy X5s core XY machine and I would really like to get this dual axis driver enabled and working. If you have a working machine, especially if it's a coreXY, could you please let me have a look at the config.h and the config_adv.h or if someone not as blonde as myself could offer some advice, I'd be eternally grateful. I can't believe I'm the only idiot out there that can't get this to work so might be worth reopening this thread.

gnattycole, 2020-01-01:
> Closing this issue because we have this very board running dual z motors without any issue on one of our printers. I will gladly reopen it if anyone begs to differ.

Hi Shitcreek, any chance you could reopen this thread until I can get a couple of replies?

jakedrost, 2020-01-18:
I'm having the same issue, the motor in my E1 slot spins twice the distance of the motor in my Z slot. I would like this to be reopened as well.

Sc0iL, 2020-01-20:
I have the same problem. 
My settings are Z2 on E0 and the extruder on E1. 
M122 spits out only errors, bad response and ALL HIGH on Z2.
I can't give you more information until tomorrow.

gnattycole, 2020-01-21:
Hi Folks,
I finally got this sorted out on my CoreXY machine. The board is a Bigtreetech SKR1.3 and I'm using TMC 2208 in UART mode. The settings I needed to use in the end were to select the Z2 motor in the config.h and not the E1. This was left disabled. In config_adv.h, I enabled dual Z and auto Z axis align. I didn't enable the dual Z endstops as I used the Z probe on the z min endstop. This got the motors spinning at the same rate but it kept crashing when I sent it a G34 command. It seemed to do the auto home ok but then it would ignore the probe triggering and crash into the head. It turns out in the end this was down to me being blonde and having my Z dir reversed. I enabled the invert Z dir and it now works like a charm. If anyone is interested, I can upload a copy of my config once I get home at the end of this week.

gnattycole, 2020-01-21:
> I have the same problem.
> My settings are Z2 on E0 and the extruder on E1.
> M122 spits out only errors, bad response and ALL HIGH on Z2.
> I can't give you more information until tomorrow.

ScOil, not sure that will work as you have it wired. My Z1 is in the normal Z socket on the board. The Z2 motor plugs in to the E1 socket and the Extruder is plugged into E0. With this setup, I didn't have to screw around with re-assigning pins and outputs. The only thing I have left to configure now are the part and hot-end cooling. I'd like a Stepper driver cooling fan but I've run out of output pins on the SKR now. Guess I'll need to upgrade to the Pro board some time in the future.

Sc0iL, 2020-01-21:
I've connect Z2 to E1.
Enable #define E1_DRIVER_TYPE TMC2130
Disable //#define Z2_DRIVER_TYPE TMC2130
The TMC connection is fixed with this config but the stepper doesn't move. 

When I click home for X or Y axis rotates the z stepper. 
But X and y are not connected, it's only a test build... 

It's a little bit creepy

gnattycole, 2020-01-22:
Hi Scoil,
I have E1 disabled and Z2 enabled in my config but there may be different requirements between the 2130 and the 2208 drivers. The Z axis might move when auto homing the X and y as there's a setting in config that  can drive to a minimum clearned before mOving X and Y axis.. 
  Here's a link to a quick video of the auto Z axis align working.  Not the best clip on the planet but hopefully shows how it's supposed to work.
https://youtu.be/Re3UDKtiwLo

Sc0iL, 2020-01-23:
@gnattycole yesterday, I've disabled E1 and enabled only Z2.
It works correctly. 

The only problem is the homing of XY, but I think this is because there is currently no endstop for XY and no stepper connected.