MarlinFirmware/Marlin #15112: [BUG] Failed to compile with Arduino IDE 1.8.9

HuguesDug, 2019-08-31:
Downloaded the latest bug fix branch 2.0 (August 30th)
Applied configuration and configuration_adv for Anet A8
Adjusted a few parameters (like PID values, Z offset, bedleveling bilinear, linear advance and slim LCD menu)
Compiled the source with Arduino IDE 1.8.9

Generate a seg fault, with following message

sketch\src\gcode\queue.cpp: In function 'gcode_line_error':
sketch\src\gcode\queue.cpp:301:1: internal compiler error: Segmentation fault
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte Anet V1.0 (Optiboot)

GMagician, 2019-08-31:
AFAIK this is a compiler error and

> Please submit a full bug report,
with preprocessed source if appropriate.
See http://gcc.gnu.org/bugs.html for instructions.

should clearify things

RudyBenner, 2019-08-31:
Compiles fine in VSCode.

solhaga12, 2019-09-02:
In queue.h try:
  static void gcode_line_error(PGM_P const err, const int8_t port) _O2;

thinkyhead, 2019-09-04:
Those hacks may work, but I am not sure we need to deploy them.

closed 2019-09-04 by thinkyhead