MarlinFirmware/Marlin #15116: tmc2130 direction change problem

hosseinvh96, 2019-08-31:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

hosseinvh96, 2019-08-31:
Description
hi im using tmc2130 driver over spi connection in marlin 2 firmware my problem is step motors only rotate cw in marlin 
Steps to Reproduce
my endstops setting are ok 
pins connected 
i used tmc2130 simple example and when it change direction over spi it works correct but i found that in marlin direction change via dir pin i want change my marlin setting to change direction via spi
Expected behavior: go step cw and ccw

Actual behavior: it goes only cw

boelle, 2019-08-31:
please attach 2 config files as the issue template says

hosseinvh96, 2019-08-31:
thanks for your reply 

[Configuration.txt]
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3562862/Configuration_adv.txt)
(https://github.com/MarlinFirmware/Marlin/files/3562861/Configuration.txt)

teemuatlut, 2019-09-02:
Why do you want to change the direction through SPI? Marlin doesn't currently support this for TMC drivers.

hosseinvh96, 2019-09-03:
how can i change direction from dir_pin to shaft.direction in marlin ?

teemuatlut, 2019-09-03:
Edit `stepper_indirection.h`.

hosseinvh96, 2019-09-03:
How can i edit it .the chage function is `st.shaft(dir)`

hosseinvh96, 2019-09-03:
i changed `  #if AXIS_IS_TMC(X)
    extern TMC_CLASS(X, X) stepperX;`
to `  #if AXIS_IS_TMC(X)
    extern TMC_CLASS(X, X) stepperX;
	#define X_DIR_WRITE(STATE) WRITE(X_DIR_PIN,STATE)`
 but noting changed

Bob-the-Kuhn, 2019-09-03:
Changing direction via SPI will take a lot of work with no apparent (to me) benefits.  Adding a SPI transaction for every direction change may slow down printing a little.

Look at the L6470 code in `stepper.cpp`, `stepper_indirection.h` and `stepper_indirection.cpp`.  L6470 does direction changes via SPI.

AnHardt, 2019-09-03:
Just to repeat that.
SPI during an interrupt needs its own SPI-buss. Else it will mess up all other SPI-transfers on that buss.

If not using LinearAdvance all direction settings are made at the end of the previous bock, followed by a relative long pause (when direction is changed we have to be slow). That should be long enough to finish the transfer before the next step comes. If using LA the extruders can change direction during a block. That could be challenging.

hosseinvh96, 2019-09-03:
but i want to know how other people use tmc2130 in spi and they dont have problem

Bob-the-Kuhn, 2019-09-03:
I'm 99.9% sure the problem is not the interface.  The most likely cause of the stepper moving in only one direction is the current setting is too low.

1) Make sure M122 doesn't report any communications problems
2) Increase the current in `Configuration_adv.h`

Diederikvanderpant, 2019-09-06:
@hosseinvh96  Today I fixed a similar problem on my own driver by soldering the "dir" trace back to the chip. The symptom of the problem was: that after months of normal operation the driver suddenly only turned in one direction. After inspection with a microscope I found some questionable solder joints on the Trinamic IC specifically the dir line.  Please pm me if you think you might have a similar problem.

boelle, 2019-09-24:
@hosseinvh96 still having problems?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle