MarlinFirmware/Marlin #15119: [ERROR]Marlin 2.0 - SKR 1.3 error: 'internalSerial' was not declared in this scope

Sythx, 2019-08-31:
### Description

error: 'internalSerial' was not declared in this scope

### Steps to Reproduce

I had before my SKR 1.3 to expand with a MMU2, when setting Marlin 2.0 (22.8 download, synonymous with the current version) appears an error. pastbin-link inserted below.

#### Additional Information
https://pastebin.com/H0AJp4ce
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3562710/Configuration.zip)
[Marlin-2.0_22.8.2019.zip](https://github.com/MarlinFirmware/Marlin/files/3562716/Marlin-2.0_22.8.2019.zip)

Ludy87, 2019-09-01:
> For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)

Sythx, 2019-09-01:
I also read this information, but I do not understand what to look for.

Sythx, 2019-09-01:
/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT_2 2

and 

  // Serial port used for communication with MMU2.
  // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
  // For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)
  #define INTERNAL_SERIAL_PORT 2
  #define MMU2_SERIAL Serial2

its work.

closed 2019-09-01 by Sythx

Sjoerrdd, 2019-12-10:
@Sythx I got the error that `MSerial2` is not defined.
Could you share your config files with us?