MarlinFirmware/Marlin #15120: [FR][2.0 (feature request title)

MarkMan0, 2019-08-31:
### Description

### Steps to Reproduce

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Grogyan, 2019-09-01:
That wastes CPU time.

Better option is to ensure the Power Loss detection pin is on an Interrupt pin.
But this is assuming that you have power loss detection circuit, such as the MKS DET, or Prusa Power Panic.

MarkMan0, 2019-09-01:
Since posting I did some changes and testing.

I created a function `PrintJobRecovery` to check the current state of the pin.

`static inline bool checkPower(){ return (READ(POWER_LOSS_PIN) != POWER_LOSS_STATE); }`

And the idle() task now looks like this:

```
if(recovery.enabled) {
      const millis_t ms = millis();
      if(ELAPSED(ms, recovery.next_save)) {
        if(recovery.checkPower() == false) {
          //no power
          //checkpower is an inline function, to avoid unnecessary overhead of function call to save()
          recovery.save();
        }
        recovery.next_save = ms + recovery.save_interval;
      }
    }
```

This should be faster than the previous proposition, if the compiler inlines the code(which it should).

I do have a detection board, which is fairly fast - 555 timer missing pulse detector.

Interrupt pin would be the ideal solution, unfortunately the interrupt capable pins on the skr 1.3 are occupied by E0 pins.

I've done a few test prints, and didn't notice anything wrong so far. But needs more testing ofc.

AnHardt, 2019-09-01:
The interrupt is a good idea, but mainly to switch off the heaters as fast as possible and then setting a flag.
Polling that flag instead the pin is about that slow/fast as polling the pin, but the much shorter reaction time, of the interrupt, safes some of the then rare energy. Halting the other processes and saving is better done in a known state than in an interrupt, 
The currently saved state is far form perfect. The contend of the planner buffer gets lost.

thinkyhead, 2019-09-02:
> The content of the planner buffer gets lost.

Rather than save the contents of the planner buffer and the current circular queue of un-processed commands, it would make the most sense to keep track of the SD position of the command at the front of the planner. While this will cost a bit of SRAM to implement, the power loss structure / file can be much smaller.

AnHardt, 2019-09-02:
> ... keep track of the SD position of the command at the front ...

The same could be done for 'N' when printing over a line. If the start-code has a M110 and commands sent manual are without a line-number, ... , so 'N' could repeatable represent a position in a sent file. Or a line, sent from a file could have a 'L'.number replacing the 'N'. 
When resuming we then could send a special answer to M110, containing the N we want the host to continue - a kind of a seek. 
However, before we discuss that with the host-team, we have to fix the basics, when printing from SD.

MarkMan0, 2019-09-02:
By the way, here are the changes I'm using right now:
[implementation](https://github.com/MarkMan0/Marlin/commit/ab373ef73bfccfa5b61d3eec2050be912ebc769d)
[removed call to save()](https://github.com/MarkMan0/Marlin/commit/d22f4a44cbf1a779225d3171cb1275462e734865)

thinkyhead, 2019-09-02:
If someone wants to extend PLR to host-based printing, that would be a fine project for the Marlin 2.1 milestone.

See #15135 for an updated PLR that keeps track of the SD file position of each command in the queue. The last commit additionally stores the SD file position in planner blocks so that PLR can resume from the beginning of the last active move.

The implementation is imperfect, but it demonstrates the gist of the thing. To fix it further, it will also need to store the starting position of the move in the planner (rather than the global `current_position` which is ahead of the planner) or —ideally— the current cartesian position according to the steppers. With that position stored, PLR could pick up in the middle of a line.

thinkyhead, 2019-09-02:
> By the way, here are the changes I'm using right now:

I see we both converged on similar code. The one thing I didn't add was a power check interval. The pin check alone seems to be faster than the logic to check whether it should be checked.

AnHardt, 2019-09-02:
That's nice progress. 
If the moves are segmented it needs further refinement (delta, scara, leveling, ...).

thinkyhead, 2019-09-04:
It's definitely going to be better than it was.

In the existing code PLR restarts shortly before the beginning of the most recent layer (because it saves based on the not-yet-executed Z position). That could be a lot of waste and makes failure more likely. If PLR is configured to save at regular intervals ("every command mode") then it will restart shortly before the last command that was *fetched*, which is better. But as pointed out, the moves in the planner will be lost forever. So either it does way too much or it skips over a little.

With the latest update, PLR will always restart at the beginning of the line currently being printed. Most of the time this will be a pretty short line. I'll put it through some heavy testing tonight.

thinkyhead, 2019-09-08:
Move discussion over to #15135…

closed 2019-09-08 by thinkyhead