MarlinFirmware/Marlin #15123: tlsv1 alert protocol version - Can't build Marlin 2.0

edwilliams16, 2019-09-01:
To install Marlin 2.0.x_bugfix on my printer, I've installed Atom and platformio on an old (2009) Mac I use to control it.  When I attempt to build Marlin, I get an SSL error in the python script that downloads stuff from github:

SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /MarlinFirmware/U8glib-HAL/archive/bugfix.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),)):

 This is apparently because Mac 10.11.6's system python doesn't support the required SSL protocol.
So installed the latest python2 and pip and put them at the beginning of my PATH so that
>%python
>Python 2.7.16 (v2.7.16:413a49145e, Mar  2 2019, 14:32:10) 
>[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
>Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> import ssl
>>> 
>>> print "Python info: %s" % (platform.python_version())
Python info: 2.7.16
>>> print "OpenSSL info: %s" % (ssl.OPENSSL_VERSION)
OpenSSL info: OpenSSL 1.0.2q  20 Nov 2018

( FYI with Apple's python:
>% /usr/bin/python
>Python 2.7.10 (default, Oct 23 2015, 19:19:21) 
>[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] on darwin
>Type "help", "copyright", "credits" or "license" for more information.
>>> import ssl
>>> print "OpenSSL info: %s" % (ssl.OPENSSL_VERSION)
>OpenSSL info: OpenSSL 0.9.8zh 14 Jan 2016
)

  but I get exactly the same errors. It appears platformio is still using the system python.
FWIW, I did the same installation on a newer Mac OSX  10.13 - everything works fine there.

Any idea how to force platformio to use the correct python or otherwise fix this problem?  I can't upgrade the MacOS, the machine won't support it.

RudyBenner, 2019-09-01:
I believe Atom no longer supports Platformio. VSCode is the other popular choice.

sl1pkn07, 2019-09-01:
Use python3 instead of python2, python2 gets EOL from the last of this year.

In macos, use homebrew for install it

Greetings

edwilliams16, 2019-09-01:
Thanks.  I already installed python3 with homebrew.  Platformio doesn't appear to use it. If I could figure out how to force it to use the right python I'd be good.  The hack of simply replacing the system python with a simlink to a good one requires trashing the security on the Mac.  The installation does work fine on a later Mac with a more recent python.
  However, I'll try VSCode since I'm not that familiar with either IDE.

sl1pkn07, 2019-09-01:
platformio from homebrew uses python3 https://formulae.brew.sh/formula/platformio

edwilliams16, 2019-09-01:
Fixing this was ridiculously complicated.  VSCode failed exactly the same way as Atom did - it's a platformio problem. To get platformio to work I had to run it in a virtual environment specifying the path to the right python following https://github.com/platformio/platformio-vscode-ide/issues/88
I'm still on python 2.7.16.  I'll figure out how to get everything on python3 later perhaps.

pip install virtualenv
rm -rf ~/.platformio/penv
/path/to/virtualenv ~/.platformio/penv -p /path/to/custom/python
chmod +x ~/.platformio/penv/bin/*
~/.platformio/penv/bin/activate
(penv) pip install platformio
(penv) pio upgrade --dev

sl1pkn07, 2019-09-02:
In my macos. I only installed platformio and vscode from homebrew/homebrew cask, and install/configure vscode PIO IDE (extension settings) for use external PIO instead of PIO core (downloaded throught vscode )

edwilliams16, 2019-09-02:
@sl1pkn07   Which MacOS?  Was a problem for El Capitan - fine on High Sierra.

sl1pkn07, 2019-09-02:
mojave

edwilliams16, 2019-09-02:
I think you can pretty much any platformio install with Mojave as the system python has the right SSL library.

sl1pkn07, 2019-09-02:
system python is not updated

edwilliams16, 2019-09-03:
> system python is not updated

Technically correct it appears - the python version is the same - however the later one is compiled against a later SSL version, compatible with github. The SSL error is why platformio failed.

MacBookPro17:ed$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.13.6
BuildVersion:	17G6030
MacBookPro17:ed$ /usr/bin/python
Python 2.7.10 (default, Oct  6 2017, 22:29:07) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ssl
>>> ssl.OPENSSL_VERSION
'LibreSSL 2.2.7'

versus

MacBookPro15:~ williams% sw_vers
ProductName:	Mac OS X
ProductVersion:	10.11.6
BuildVersion:	15G22010
MacBookPro15:~ williams% /usr/bin/python
Python 2.7.10 (default, Oct 23 2015, 19:19:21) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ssl
>>> ssl.OPENSSL_VERSION
'OpenSSL 0.9.8zh 14 Jan 2016'

Roxy-3D, 2019-09-03:
@edwilliams16     For the record...   I'm not a big fan of PlatformIO.   It is unstable as $#/+.

edwilliams16, 2019-09-03:
@Roxy-3D 
>For the record... I'm not a big fan of PlatformIO. It is unstable as $#/+. 

Doesn't seem like users have much choice if they want to go 32-bit. http://marlinfw.org/docs/basics/install.html

What would have helped me the most while spelunking the code this week would have been some preprocessor tool that would indicate what lines/files in the code are actually active in a given configuration.  Has anyone looked into anything like coan  http://coan2.sourceforge.net/
 Maybe it could help even seasoned developers who know their way around the code.

sl1pkn07, 2019-09-03:
~~~
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  sw_vers
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G87
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  python
Python 3.7.4 (default, Jul  9 2019, 18:13:23) 
[Clang 10.0.1 (clang-1001.0.46.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>>
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  which python
/usr/local/opt/python/libexec/bin/python
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  /usr/bin/python
Python 2.7.10 (default, Feb 22 2019, 21:55:15) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.1 (clang-1001.0.37.14)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> ^D
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  cat .bash_profile
---snip---
export PATH="/usr/local/opt/python/libexec/bin:$PATH"
---snip---
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  brew info platformio
platformio: stable 4.0.3 (bottled)
Ecosystem for IoT development (Arduino and ARM mbed compatible)
https://platformio.org/
/usr/local/Cellar/platformio/4.0.0 (1,124 files, 11.5MB) *
  Poured from bottle on 2019-08-11 at 19:24:37
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/platformio.rb
==> Dependencies
Required: python ✔
==> Analytics
install: 962 (30 days), 2,030 (90 days), 8,314 (365 days)
install_on_request: 919 (30 days), 1,940 (90 days), 7,725 (365 days)
build_error: 0 (30 days
~~~

with not updated, i mean macos use python2 (EOL from the end of this year) instead of python3

sl1pkn07, 2019-09-03:
if you like, also can install the latest(?) python2 version throught brew

~~~
┌─┤[$]|[Hakintoch]|[Hackintoch]|[~]|
└───╼  brew info python@2
python@2: stable 2.7.16 (bottled), HEAD
Interpreted, interactive, object-oriented programming language
https://www.python.org/
Not installed
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/python@2.rb
==> Dependencies
Build: pkg-config ✔
Required: gdbm ✔, openssl@1.1 ✔, readline ✔, sqlite ✔
==> Options
--HEAD
	Install HEAD version
==> Caveats
Pip and setuptools have been installed. To update them
  pip install --upgrade pip setuptools

You can install Python packages with
  pip install <package>

They will install into the site-package directory
  /usr/local/lib/python2.7/site-packages

See: https://docs.brew.sh/Homebrew-and-Python
==> Analytics
install: 87,318 (30 days), 266,418 (90 days), 1,694,124 (365 days)
install_on_request: 25,469 (30 days), 77,652 (90 days), 418,628 (365 days)
build_error: 0 (30 days)
~~~

edwilliams16, 2019-09-03:
If you read my initial message, you'll find that I did the obvious thing and installed the latest python2 (and python3) and put them to the head of my path - trouble was platformio ignores the user's $PATH and uses the system python on install  The work-around described above was to install platformio into a virtual environment with the appropriate python.  
  I have no experience with hackintoshes - can you upgrade, say /usrbin/python without disabling system integrity protection - you can't on a real Mac.

sl1pkn07, 2019-09-04:
/usr/bin is untouched. homebrew(brew) uses /usr/local. if you want use the executables installed in it, you need add the PATH environment variable in your bash_profile (in you user, not in the root)

in my "mac" (real machine, no VM), i first i install homebrew (see the homebrew docupentation), then 

>brew install platformio

brew install the dependencies iself, include the "correct" python, then add the line 

>export PATH="/usr/local/opt/python/libexec/bin:$PATH"

in you `.bash_profile` in your user home.
-logout and login again

Add the Cask repository

>brew tap caskroom/cask

install VScode

>brew cask install visual-studio-code

open the vscode and go Extension manager

- install platformio IDE

and go to platformio IDE settings. or general settings and search "platformio"

- Unclick: use built-in platformio core
- Unclick: update terminal configuration with patched PATH environment

and go

edwilliams16, 2019-09-04:
OK.  Did all those things in various orders. Maybe it works. My experience and that in https://github.com/platformio/platformio-vscode-ide/issues/88
suggests it might not.  But further readers can try both approaches.

boelle, 2019-09-24:
@edwilliams16 so you can compile/build ?

edwilliams16, 2019-09-24:
I can compile/build.  No problem with Marllin - only with the platformio Mac install - where I found a work-around.

boelle, 2019-09-24:
oki doke, will close then as its not a marlin bug

closed 2019-09-24 by boelle