MarlinFirmware/Marlin #15133: getting 3dtouch working with SKR-mini E3

Rene046, 2019-09-01:
First post here did not know if im right here.

I connected my 3dtouch to the servo and probe connectors onto the SKR-mini 3E al wires polarization like it should be.

Activated in marlin configuration.h 
#define BLTOUCH
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

uploaded bin file true sd card
3dtouch does test function when started, and probe up and down work also,
but if i home the printer the z axis only stops with the Z end switch not the probe.

did i forget something

thisiskeithb, 2019-09-01:
Please follow the Issue template:

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

Rene046, 2019-09-02:
thx i added it to the first post 
could be that i disabled in the Configuration.h the #define BLTOUCH
because i was experimenting....

thanks in advance for any help.

LPRM, 2019-09-02:
I got 3DTouch working with the files included in the zip.
[3dtouch7points.zip](https://github.com/MarlinFirmware/Marlin/files/3567176/3dtouch7points.zip)

Rene046, 2019-09-03:
Hi Luís Pedro Morais

i compared your files with mine and changed some stuff like you have.
Now printer is homing in center. but probe still won't make my z axis stop  only my end switch of the zaxis in octoprint i get following message:
Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: STOP called because of BLTouch error - restart with M999

Strange if i want to use M401 or M402 true octoprint terminal probe does nothing, if i use octoprint command buttons it goes up and down.
If i type M119 with pin down i get 
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: y_min: TRIGGERED
Recv: z_min: open
And with the pin UP i get the same and BL pin shoots down ?
With pin up my led is red/blue and pin down it's blue... both position i read 0 volts over the black and white cable.
If pin is down and i push it for a few seconds up it's starts to blink and i measure 5 volt.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3570705/Marlin.zip)

LPRM, 2019-09-03:
I suggest you check out this: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues/6
Follow the steps written by the user sky8000 and uncomment `#define BLTOUCH_FORCE_SW_MODE`.
I connected the 3DTouch to "SERVO" and "PROBE" pins.

shitcreek, 2019-09-03:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-09-03 by shitcreek