MarlinFirmware/Marlin #15134: error LULZBOT_TOUCH_UI

Lino77, 2019-09-01:
It may be that a few configuration settings are missing. I can not find MAX_MANUAL_FEEDRATE anywhere

tmp/arduino_build_721483/sketch/src/lcd/extensible_ui/lib/lulzbot/screens/move_axis_screen.cpp: In static member function 'static float MoveAxisScreen::getManualFeedrate(uint8_t, float)':
/tmp/arduino_build_721483/sketch/src/lcd/extensible_ui/lib/lulzbot/screens/move_axis_screen.cpp:112:47: error: 'MAX_MANUAL_FEEDRATE' was not declared in this scope
   constexpr float max_manual_feedrate[XYZE] = MAX_MANUAL_FEEDRATE;
                                               ^
Bibliothek Wire in Version 1.0 im Ordner: /home/lino/.arduino15/packages/ultimachine/hardware/sam/1.6.9-c/libraries/Wire  wird verwendet
Bibliothek TMCStepper in Version 0.4.6 im Ordner: /home/lino/Arduino/libraries/TMCStepper  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /home/lino/.arduino15/packages/ultimachine/hardware/sam/1.6.9-c/libraries/SPI  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Archim.

boelle, 2019-09-02:
attach 2 config files as zip file just as the issue template suggests

Lino77, 2019-09-02:
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/3567088/Marlin_config.zip)

Ludy87, 2019-09-04:
Das sollte nicht auskommentiert sein und dann muss es ein Array sein
https://github.com/MarlinFirmware/Marlin/blob/d9bbe0ff02530239dfa726b87e8303ff427755ba/Marlin/src/lcd/extensible_ui/lib/lulzbot/config.h#L93

#define MAX_MANUAL_FEEDRATE { 240, 240, 240, 240 }

Aber ich denke, da werden noch mehr Fehler auftauchen.

@marcio-ao can you say more about that?

marcio-ao, 2019-09-05:
@Ludy87: Yes, it looks like that config file is incorrect. I agree that the following would be a fix:

`#define MAX_MANUAL_FEEDRATE { 240, 240, 240, 240 }`

I'll create a PR with this fix.

Ludy87, 2019-09-05:
Thx @marcio-ao

Lino77, 2019-09-06:
@Ludy87 hattest recht es kam noch ein Fehler anbei meine Config und was ich geändert habe das es durch kompiliert jetzt fehlt mir nur noch das Display zum testen hoffentlich kommt es bald, sollte schon letzte Woche eigentlich da sein, kannst du mir sagen für was "LCD_STR_THERMOMETER" eigentlich steht 

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3581920/Config.zip)

Ludy87, 2019-09-06:
that's because it's not called `HAS_GRAPHICAL_LCD` || `HAS_CHARACTER_LCD` is declared

marcio-ao, 2019-09-13:
@Lino77: For the moment, you should be able to bypass the compilation error by disabling TOUCH_UI_USE_UTF8 in `Configuration_adv.h`. I'll work on a pull request with a fix.

As for why your display is blank, uncomment "#define UI_FRAMEWORK_DEBUG" in [Marlin/src/lcd/extensible_ui/lib/lulzbot/config.h](https://github.com/MarlinFirmware/Marlin/blob/f9961ddebb125cb6cc6670b82093d25ee4477f76/Marlin/src/lcd/extensible_ui/lib/lulzbot/config.h#L93)

Then recompile. Then power on the printer and immediately connect via serial. You can confirm whether the USB is working by looking for the following:

```
Testing X connection... OK␊
Testing Y connection... OK␊
Testing Z connection... OK␊
Testing E connection... OK␊
Testing E1 connection... OK␊
echo:Saving DL to RAMG cache, bytes: 1368 Free space: 2048␊
echo:New status message: LulzBot TAZ Pro ready.␊
echo:New screen: 1␊
echo:SD init fail␊
echo:Active Extruder: 0␊
Starting USB host...rev.03 started␊
echo:Active Extruder: 0␊
X:-27.00 Y:292.00 Z:290.00 E:0.00 Count X:-2700 Y:29200 Z:145000␊
ok P15 B5␊
echo:New screen: 3␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
echo:Appending to DL from RAMG cache, bytes: 1368 REG_CMD_DL: 1368␊
```

The important bit is to look for ```Starting USB host...rev.03 started``` and that you are seeing periodic ```Appending to DL from RAMG cache```. The numbers may be slightly different.

Unfortunately, the way the FT810 works, the screen is blank even if it is powered, so troubleshooting can be a bit tricky. We first need to make sure communications is happening before we can move to the next steps.

Lino77, 2019-09-13:
@marcio-ao 
with inserted FT810 my Archim2 does not start, without see log
[log.txt](https://github.com/MarlinFirmware/Marlin/files/3611593/log.txt)

marcio-ao, 2019-09-13:
@Lino77: Something isn't quite right. Check your wiring. You could start by disabling the `LULZBOT_TOUCH_UI` and making sure the USB is starting up. If you get `Starting USB host...rev.03 started` then you at least would know the GND, 5V, MISO, MOSI and CLK, CS_N are correct.

Lino77, 2019-09-13:
The USB port works only with activated Touch_ui. look at the times the log.txt, the Usb Port starts yes, but the rest does not want it to be that I have a broken ft810 I can not test the part somehow

Drucker ist jetzt Online.

echo:Not enough space in GRAM to cache display list, free space: 2048 Required: 8191
echo:CachedScreen::storeBackground() failed: not enough DL cache space
echo:New status message: 3D Printer Ready.
Faulted... ignoring write.
Faulted... ignoring write.
echo:Appending to DL from RAMG cache, bytes: 65535 REG_CMD_DL: 4294967295
Faulted... ignoring write.
echo:Waiting for 4 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
echo:Waiting for 4 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Starting USB host...rev.03 started
Faulted... ignoring write.
Faulted... ignoring write.
echo:Appending to DL from RAMG cache, bytes: 65535 REG_CMD_DL: 4294967295
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.

marcio-ao, 2019-09-13:
With the Achim board, it is difficult to see error messages at startup, as the board always resets the serial connection.

So, edit "Marlin/src/lcd/extensible_ui/lib/lulzbot/ftdi_eve_lib/basic/commands.cpp" and replace:

```
  /* read the device-id until it returns 0x7c or times out, should take less than 150ms */
  uint8_t counter;
  for(counter = 0; counter < 250; counter++) {
   uint8_t device_id = mem_read_8(REG::ID);            // Read Device ID, Should Be 0x7C;
   if (device_id == 0x7c) {
     #ifdef UI_FRAMEWORK_DEBUG
       SERIAL_ECHO_START();
       SERIAL_ECHOLNPGM("FTDI chip initialized ");
     #endif
     break;
   }
   else {
     delay(1);
   }
   if (counter == 249) {
     #ifdef UI_FRAMEWORK_DEBUG
       SERIAL_ECHO_START();
       SERIAL_ECHOLNPAIR("Timeout waiting for device ID, should be 124, got ", device_id);
     #endif
   }
  }
```

With:

```
  /* read the device-id until it returns 0x7c or times out, should take less than 150ms */
  uint8_t counter;
  for(counter = 0; counter < 250; counter++) {
   uint8_t device_id = mem_read_8(REG::ID);            // Read Device ID, Should Be 0x7C;
   if (device_id == 0x7c) {
     for(;;) {
         SERIAL_ECHO_START();
         SERIAL_ECHOLNPGM("FTDI chip initialized ");
         safe_delay(1000);
     }
     break;
   }
   else {
     delay(1);
   }
   if (counter == 249) {
      for(;;) {
         SERIAL_ECHO_START();
         SERIAL_ECHOLNPAIR("Timeout waiting for device ID, should be 124, got ", device_id);
         safe_delay(1000);
      }
   }
  }
```

This should cause your printer to hang at startup printing either `FTDI chip initialized` or `Timeout waiting for device ID, should be 124, got ...`.

marcio-ao, 2019-09-13:
Also, let me know exactly which display you purchased.

marcio-ao, 2019-09-13:
Another idea: Power on the printer with your finger touching the display. This is a special condition that will tell the printer it to force the settings to default. Sometimes the LCD brightness could be stuck at 0% you won't be able to see anything unless you force the settings to the defaults.

Lino77, 2019-09-13:
this display 
http://www.hotmcu.com/5-graphical-lcd-touchscreen-800x480-spi-ft810-p-286.html

Lino77, 2019-09-13:
[image]

Lino77, 2019-09-13:
with the connected display. does not boot the board

marcio-ao, 2019-09-13:
Check your pinout. I noticed that when you posted your pin mappings, the arrangement of the pin numbers was opposite of the ones [here](https://github.com/MarlinFirmware/Marlin/blob/f9961ddebb125cb6cc6670b82093d25ee4477f76/Marlin/src/lcd/extensible_ui/lib/lulzbot/pin_mappings.h#L111)

Also, use a multimeter and make sure you can read 5V between the 5V and GND pads next to the ribbon cable on the FTDI board.

Lino77, 2019-09-13:
That was because I kept to the assignment of Ultimachine. pin assignment is correct 5v are at the board anse.sehe I really the new 5V GND MOSI MISO SCK CS PD needed
INT and audio are free
[image]

Lino77, 2019-09-14:
Today's Marlin, it's just touch ui and USB support enabled,

/tmp/arduino_build_204641/sketch/src/lcd/extensible_ui/lib/lulzbot/screens/boot_screen.cpp: In static member function 'static void BootScreen::onIdle()':
/tmp/arduino_build_204641/sketch/src/lcd/extensible_ui/lib/lulzbot/screens/boot_screen.cpp:65:7: error: 'loadBitmaps' is not a member of 'StatusScreen'
       StatusScreen::loadBitmaps();
       ^
/tmp/arduino_build_204641/sketch/src/lcd/extensible_ui/lib/lulzbot/screens/boot_screen.cpp:78:5: error: 'loadBitmaps' is not a member of 'StatusScreen'
     StatusScreen::loadBitmaps();
     ^

Lino77, 2019-09-17:
Okay, I have omitted the touch, and a normal smart controller connected
USB works fine.

Connecting ...
Printer is online now.
USB device inserted
LUN is good
echo: SD card ok
I think my FT810 is broken.
Can I test the FT810 in any way?
every time I plug in the display I lose the connection to Pronterface and nothing works anymore

marcio-ao, 2019-09-17:
@Lino77: Try connecting a few pins at a time. First connect GND and 5V and check whether the board reboots. Then connect PD, CS, SCLK, MOSI and MISO in that order. See where the board starts to reboot. This will give you information on where the problem is.

Lino77, 2019-09-19:
@marcio-ao 
GND 5V OKAY 
PD okay 
CS Okay 
SCLK Okay 
Mosi OKay 
MISO not okay Printer hangs  

echo:Waiting for 16 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
echo:Touch start: 3
echo:Appending to DL from RAMG cache, bytes: 65535 REG_CMD_DL: 57373
echo:Waiting for 4 bytes in command queue, now free: 0... done
echo:Waiting for 12 bytes in command queue, now free: 0... done
echo:Waiting for 12 bytes in command queue, now free: 0... fault
echo:Waiting for 12 bytes in command queue, now free: 0... done
echo:Waiting for 4 bytes in command queue, now free: 0... fault
echo:Waiting for 12 bytes in command queue, now free: 0
[ERROR] Can't read from printer (disconnected?) (SerialException): device reports readiness to read but returned no data (device disconnected or multiple access on port?)
[ERROR] Can't write to printer (disconnected?) (SerialException): write failed: [Errno 5] Input/output error

marcio-ao, 2019-09-19:
@Lino77: So MISO is "Master In, Slave Out", meaning it is a signal that comes *from* the LCD module *to* the Archim board. The fact that your board is crashing when that is connected indicates a couple possibilities:

1) Both sides are trying to drive a signal into the pin. This will cause a short which could cause the board to crash.
2) It is possible the pin on the LCD side is being driven at 5V, while the pins on the Archim are only 3.3V tolerant (this is unlikely)

If you have a resistor available (something in the order of 1kOhm), try to put that in series with the MISO line. If the board ceases to crash, then we know whether these issues were at play. Once we get the board not to crash, we can continue troubleshooting.

Lino77, 2019-09-19:
1kOhm resistance is in it. the board does not crash anymore, but still black

echo:Waiting for 4 bytes in command queue, now free: 0... done
echo:Saving DL to RAMG cache, bytes: 0 Free space: 1048576
echo:Waiting for 16 bytes in command queue, now free: 0... done
echo:Waiting for 4 bytes in command queue, now free: 0... done
echo:Waiting for 12 bytes in command queue, now free: 0... done
echo:Waiting for 12 bytes in command queue, now free: 0... done
echo:Waiting for 12 bytes in command queue, now free: 0... done
Faulted... ignoring write.
echo:Waiting for 12 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
echo:Waiting for 12 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
echo:Waiting for 12 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
echo:Waiting for 4 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.

marcio-ao, 2019-09-19:
@Lino77: Okay, that's good news. I guess we should confirm that the MISO pin was not damaged on the Archim.

My understanding is that you had gotten the USB to work before? Can you confirm it still works? Since the USB uses MISO, we can confirm that the MISO is good if the USB functions.

marcio-ao, 2019-09-19:
First try the USB without the display connected. Then try it with the display connected. If the first case works, then we know the MISO is good on the Archim. If connecting the display causes the USB to fail, then we know the MISO on the display is interfering somehow.

Lino77, 2019-09-19:
USB works without touch 

USB device inserted
echo:Playing note 60, instrument 86
echo:Touch end: 0
echo:Waiting for 4 bytes in command queue, now free: 0... fault
echo:Saving DL to RAMG cache, bytes: 8191 Free space: 1048577
Faulted... ignoring write.
Faulted... ignoring write.
echo:Waiting for 12 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
echo:Waiting for 12 bytes in command queue, now free: 0... fault
echo:Waiting for 12 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
Faulted... ignoring write.
echo:Waiting for 16 bytes in command queue, now free: 0... fault
Faulted... ignoring write.
Faulted... ignoring write.
LUN is good
echo:SD card ok

Lino77, 2019-09-19:
The display lights up white without the USB Reader plugged in

Lino77, 2019-09-19:
[image]

marcio-ao, 2019-09-20:
@Lino77: This is peculiar. Check one thing for me. If you disconnect MOSI, the display will remain blank, correct? The fact that it is turning white suggests to me it at least is sending commands to the display and activating the backlight.

It seems like you may have wired everything up like our TAZ Pro, so try our official FW and see if it shows anything on the screen:

[Marlin_Quiver_TAZPro_Quiver_DualExtruder_2.0.0.165_d1611cd53.bin.txt](https://github.com/MarlinFirmware/Marlin/files/3637323/Marlin_Quiver_TAZPro_Quiver_DualExtruder_2.0.0.165_d1611cd53.bin.txt)

marcio-ao, 2019-09-20:
Also, please check your Archim 2.0 for identifying information. Maybe Ultimachine made different versions of the Archim. Does it have a version number?

Lino77, 2019-09-21:
Archim V2.2b
Your Firmware works
[image]
[image]
[image]

Lino77, 2019-09-23:
@marcio-ao 
can you explain this, how do you compile your firmware

marcio-ao, 2019-09-23:
@Lino77: Excellent! Having working FW will make it much easier for us to understand what is different from upstream Marlin.

The exact firmware you are using hasn't been pushed to our repo quite yet, but I'll give you instructions on how to compile what is there right now.

```
git clone https://code.alephobjects.com/diffusion/MARLIN/marlin.git
cd Marlin
git checkout devel
```

With Linux:
=========

`./build-lulzbot-firmware Quiver_TAZPro Quiver_DualExtruder`

With Arduino IDE:
==============

Run:

`cp config/examples/AlephObjects/Quiver_TAZPro/CecropiaSilk_SingleExtruderAeroV2/* Marlin`/

Run Arduino IDE (you need the beta build, earlier ones fail) on "Marlin/Marlin.ino":

Choose "Preferences" from the "File" menu and add "https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json" to the "Additional Boards Manager URLs".

In the "Boards Manager":
    - For TAZ Pro: Search for "Archim" and install "Archim by UltiMachine"
    - For all others: Search for "RAMBo" and install "RepRap Arduino-compatible Mother Board (RAMBo) by UltiMachine"

Choose the board corresponding to your printer from the "Board" submenu menu of the "Tools" menu. For a TAZ Pro, select "Archim"; for all others, select "RAMBo"

Lino77, 2019-09-23:
okay under linux I could build a firmware and it works too. under Arduino 1.9 beta I get the following error
In file included from /tmp/arduino_build_807345/sketch/src/HAL/HAL_DUE/../../core/../inc/MarlinConfig.h:28:0,
                 from /tmp/arduino_build_807345/sketch/src/HAL/HAL_DUE/../../core/serial.h:24,
                 from /tmp/arduino_build_807345/sketch/src/HAL/HAL_DUE/DebugMonitor.cpp:26:
/tmp/arduino_build_807345/sketch/src/HAL/HAL_DUE/../../core/../inc/MarlinConfigPre.h:49:35: fatal error: ../src/Lulzbot_Extras.h: No such file or directory
 #include "../src/Lulzbot_Extras.h"
                                   ^
compilation terminated.
exit status 1

Lino77, 2019-09-23:
I have the error arduino does not transfer the firmware properly to the board. I have created a hexfile under Arduino IDE, and loaded with cura the firmeware. And lo and behold! It works

boelle, 2019-10-12:
@marcio-ao & @Lino77 any news on this one?

InsanityAutomation, 2019-10-15:
@boelle  @shitscreek this one can get closed. Arduino IDE issue.

boelle, 2019-10-15:
oki, will swing the axe then :-D

commit 47d19bab (2019-09-11): Lulzbot ExtUI Western language support (#15208)

Fix #15134
closed 2019-10-15 by boelle