MarlinFirmware/Marlin #15137: Lacking Configuration.h and another 2 GPL violations Creative3D and Twotrees

escknx, 2019-09-02:
Purchased new Creative3D Elf Corexy printer that comes with MKS Robin Nano and is advertised as Marlin 32 bit firmware. 

All I could get from them are binaries and robin_nano_cfg.txt file that allows some adjustment without compiling.

They are not the only ones that do that. I've noticed from Twotrees Sapphire facebook group that their new corexy named Sapphire Pro comes with same MKS Robin Nano board, and they supply only binaries.

All my requests to seller for configuration.h are ignored. I am unable to get BLTouch working or thermistor change.

Is there any way to recreate config.h from this robin_nano_cfg.txt?

AnHardt, 2019-09-02:
Maybe.

But without appending the file or providing a link a definitive answer is impossible. Don't expect us to search for that.

escknx, 2019-09-02:
I am sorry, my bad.
Here's the binaries and separate config.txt translated to English. Original has Chinese descriptions.
https://github.com/escknx/Creative3D-Elf-Binaries

They use 2x Z sensors for auto level and I suspect these 2 lines of code is what they are trying to shamefully protect.
https://www.youtube.com/watch?v=GDGWcwDOVtA
I communicate directly with this guy Jacky Rao on Facebook, he seems to be their boss, and he keeps saying "okay tomorrow" for the past week, still nothing happens.

boelle, 2019-09-06:
will close this one, not that it matters, but on the other hand there is not much we can do about china sellers, we can only point people to what brands and sellers that do comply with the GPL

closed 2019-09-06 by boelle