MarlinFirmware/Marlin #15138: [BUG] Printer freezes after warming up the bed

Bergerac56, 2019-09-02:
With version e8dd41ec7c6c20e7ae16823cd7825894fb126b7d (which compiles fine with PIO) the printer freezes after having reached the requested bed  temperature and at the moment it starts to heat the nozzle.

With commit a67830bf678e97391bfab1dc0aa2737fb9d71f61, which is the last level of marlin I tested before leaving for some days out, everything was normal. No freeze.

I did not have the time to track the potential suspected commit, but a test done at the level of  commit 318356b6bc7c18e2814817673541c4b3767532d4 seems to already show the issue

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3566511/Config.zip)

pa0los, 2019-09-02:
Same here on Ender3 when build from commit 9958f5e3229e7823344ef07005fbc079aa8a68ca
I have reverted to yesterday's commit: be051c890f7cb167491b49b43a2dce9f76c8856c and seems to be fine. My primary suspect is commit: 451a942e951d7802342f94d97c4013319c240296

fldc, 2019-09-02:
Ugh, I've just spent all day debugging my my newly installed SKR 1.3, thinking that was the issue, reverting commit 451a942 fixes the issue.

jinantha, 2019-09-02:
Hi if anyone can help me here. (tu)
I've changed my Ender 3 stock board to a SKR V1.3 with TMC2130 drivers. 
Now I'm unable to print anything. 
Printer just halts around 1st or 2nd layer.
And no response for any of the GCODE commands.
I had to power cycle.
 
My hardware;
Ender 3, 
BigTreeTech SKR V1.3, 
TMS2130 - BigTreeTech v3, 
BigTreeTech SKR TFT35 V1.2 

latest Marlin 2.0 bug fix (downloaded yesterday 2019-Sep-01). 

Config files, pictures and compiled firmware -> [https://www.dropbox.com/sh/vhlia1qlqtfvovi/AACfHDK3q7LXMsKnw1vZlpQ9a?dl=0](url)

I tried auto PID no change.

Bergerac56, 2019-09-02:
@pa0los @fldc I confirm. Reversing the commit solves the issue. 

@thinkyhead Hello Scott. It seems that we have an issue with commit 451a942e951d7802342f94d97c4013319c240296

Bergerac56, 2019-09-02:
@jinantha Hello. As your printer starts to print, I do not think that you have the same issue. To be sure, try reverting the commit 451a942e951d7802342f94d97c4013319c240296 as indicated above. But I have some doubts...

It could be better to open a new point to seek help

profknivre, 2019-09-02:
I can also confirm, dropping 451a942 fixes the issue for me.

jinantha, 2019-09-02:
> @jinantha Hello. As your printer starts to print, I do not think that you have the same issue. To be sure, try reverting the commit [451a942](https://github.com/MarlinFirmware/Marlin/commit/451a942e951d7802342f94d97c4013319c240296) as indicated above. But I have some doubts...
> 
> It could be better to open a new point to seek help

Thanks man will try reverting and if that do not help, i'll open a new point.

Powerprobot, 2019-09-02:
> 
> 
> With version [e8dd41e](https://github.com/MarlinFirmware/Marlin/commit/e8dd41ec7c6c20e7ae16823cd7825894fb126b7d) (which compiles fine with PIO) the printer freezes after having reached the requested bed temperature and at the moment it starts to heat the nozzle.
> 
> With commit [a67830b](https://github.com/MarlinFirmware/Marlin/commit/a67830bf678e97391bfab1dc0aa2737fb9d71f61), which is the last level of marlin I tested before leaving for some days out, everything was normal. No freeze.
> 
> I did not have the time to track the potential suspected commit, but a test done at the level of commit [318356b](https://github.com/MarlinFirmware/Marlin/commit/318356b6bc7c18e2814817673541c4b3767532d4) seems to already show the issue
> 
> [Config.zip](https://github.com/MarlinFirmware/Marlin/files/3566511/Config.zip)

The Same here:
My hardware;
Alfawise U30 complete ReWork,
BigTreeTech SKR V1.3,
TMC2208 - BigTreeTech v2 UART
BigTreeTech SKR TFT35 V1.2
BigTreeTech Full Graphics Display 12864.

latest Marlin 2.0 bug fix (downloaded today 2019-Sep-02).

tpruvot, 2019-09-02:
i can confirm the revert of 451a942 is required on the U20 too

thisiskeithb, 2019-09-02:
I'll add to this and say "me too" on an SKR 1.3'd Ender-3.

Reverting https://github.com/MarlinFirmware/Marlin/commit/451a942e951d7802342f94d97c4013319c240296 fixes the issue.

thinkyhead, 2019-09-03:
Reverted an experimental patch that … fell into … `bugfix-2.0.x`.

closed 2019-09-03 by thinkyhead