MarlinFirmware/Marlin #15141: [BUG] #15113 breaks CR10_STOCKDISPLAY support

vincentbaeten, 2019-09-02:
### Description

Today I wanted to flash the latest firmware of bugfix-2.0.x on my skr mini e3 but after flashing, only the backlight will turn on and no status screen will show up. After a bit of testing, I've found that the culprit is [#15113](https://github.com/MarlinFirmware/Marlin/pull/15113).

### Steps to Reproduce

1. Flash [#15113](https://github.com/MarlinFirmware/Marlin/pull/15113) 

**Expected behavior:** [What you expect to happen]

Normal status screen

**Actual behavior:** [What actually happens]

No status screen.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3566876/config.zip)
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-09-02:
had the same today,
had to remove eeprom emulation https://github.com/MarlinFirmware/Marlin/pull/15126
and the commit for https://github.com/MarlinFirmware/Marlin/issues/13550 broke my skr mini e3

thisiskeithb, 2019-09-02:
I just flashed my Ender-3 with `CR10_STOCKDISPLAY` and don't have this issue.

Have you tried cleaning your libraries (click the trash can icon in your IDE or manually delete the .pio folder)?

reloxx13, 2019-09-02:
> 
> 
> I just flashed my Ender-3 with `CR10_STOCKDISPLAY` and don't have this issue.
> 
> Have you tried cleaning your libraries (click the trash can icon in your IDE or manually delete the .pio folder)?

SKR MINI E3 Board ???

thisiskeithb, 2019-09-02:
> Do u have the SKR MINI E3 Board ???

No. I am using an SKR 1.3. I was going off the title assuming OP meant that `CR10_STOCKDISPLAY` was broken.

It looks like this just affects the SKR Mini E3 so far.

vincentbaeten, 2019-09-02:
> > Do u have the SKR MINI E3 Board ???
> 
> No. I am using an SKR 1.3. I was going off the title assuming OP meant that `CR10_STOCKDISPLAY` was broken.
> 
> It looks like this just affects the SKR Mini E3 so far.

Sorry it is a bit misleading indeed. I've changed the title to be more specific.

> had the same today,
> 
> 1. had to remove eeprom emulation #15126
> 2. and the commit for #13550 (this one is before ur 15113, r u sure its that?)
> 
> both broke my skr mini e3
> 
> edit:
> to 1., note u dont have eeprom settings enabled in ur config
> to 2., tested again, it does not work with commit [23bb40d](https://github.com/MarlinFirmware/Marlin/commit/23bb40d0ea398cee3f68bd9a4a716a3b7dda8db9) #13550

It is working for me on commit [23bb40de]

thisiskeithb, 2019-09-02:
I just tried `CR10_STOCKDISPLAY` and `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` on my SKR Mini 1.1 (non E3 or DIP version) and neither work with https://github.com/MarlinFirmware/Marlin/commit/12c595c61f4b3e0be2baa6a906718e155a797ea3.

shitcreek, 2019-09-03:
Got the same issue with 'REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER' and BTT SKR V1.3, but it works with 'CR10_STOCKDISPLAY'

thinkyhead, 2019-09-03:
Yeah, with the `BIGTREE_SKR_1_3` you can plug in a RRD FGSC and the pins defined for the `CR10_STOCKDISPLAY` work for it. So, maybe the condition in that pins file should be…
```cpp
#if EITHER(CR10_STOCKDISPLAY, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
```

zombu2, 2019-09-04:
Seems it's related to boards with STM mcu only ran into same issue then tried various other mcu's none showed the issue of no display

thisiskeithb, 2019-09-04:
> Seems it's related to boards with STM mcu only

`BOARD_BIGTREE_SKR_V1_3` is `LPC1768`. So maybe it's 32-bit? I only have `LPC1768` and `STM32F1` to test with.

I've been playing around with a MKS GEN-L and various 12864-type LCDs the past few nights and `Mega2560` is unaffected.

ManuelMcLure, 2019-09-04:
My Re-ARM works fine with the RRD Full Graphics Smart Controller and the latest bugfix-2.0.x sources, so it’s not a basic 32 bit problem.

zombu2, 2019-09-04:
swap out these files in /Marlin/src/lcd/dogm/ from an older tree
ultralcd_st7920_u8glib_rrd_AVR.cpp
ultralcd_st7920_u8glib_rrd_AVR.h
u8g_dev_st7920_128x64_HAL.cpp

that will get the display working again

thisiskeithb, 2019-09-04:
> swap out these files in /Marlin/src/lcd/dogm/ from an older tree
> ultralcd_st7920_u8glib_rrd_AVR.cpp
> ultralcd_st7920_u8glib_rrd_AVR.h
> u8g_dev_st7920_128x64_HAL.cpp
> 
> that will get the display working again

Yes, that was part of 12c595c, but I don't understand the changes, so I don't know how to fix it 😄

zombu2, 2019-09-04:
well a workaround is to drop older files in before the change which will make the display work

tpruvot, 2019-09-05:
its from PR #15113 we badly supposed the AVR.cpp file was specific to AVR platform, but both compile on STM32 (F4 at least, maybe F1 too for this RRD)

zombu2, 2019-09-05:
skr mini too not sure what mcu that uses

tpruvot, 2019-09-05:
mini is STM32F103R (64pins)

AbelCS, 2019-09-05:
Same problem here with E3 DIP, researching a bit I've reached 12c595c and reviewed the code, but too much u8g for me...
 
Maybe @GMagician can help

GMagician, 2019-09-05:
The only changes done that may affect other boards than samd51 are the ones in:
`arlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h`, `Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp` that changed lines 26-27, 30 and 154-157
I can't see nothing else in 12c595c

tpruvot, 2019-09-05:
just need to change the top ifdef in the 2 cpp files

tpruvot, 2019-09-08:
Thanks to confirm if its resolved...

reloxx13, 2019-09-08:
updated yesterday and it work if TMC_DEBUG is disabled.

Wont boot with TMC_DEBUG enabled.

AbelCS, 2019-09-09:
> updated yesterday and it works if TMC_DEBUG is disabled.
> 
> Wont boot with TMC_DEBUG enabled.

What board are you using? 

Works well for me with TMC_DEBUG enabled on SKR E3 DIP.

reloxx13, 2019-09-09:
skr mini e3

zombu2, 2019-09-09:
did not try it on a mini but no issues on skr 1.3 or skr pro 
not sure how tmc debug could be related to display o.0

reloxx13, 2019-09-09:
note: TMC_DEBUG worked before adding the EEPROM emulation

thisiskeithb, 2019-09-09:
> Thanks to confirm if its resolved...

I'm still getting a blank screen with `CR10_STOCKDISPLAY` on an SKR Mini 1.1 using the latest commit (841c8ce). [Link to the test branch I'm using.](https://github.com/thisiskeithb/Marlin/tree/skr-mini-1.1-testing)

`REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` (native & emulated) work.

No issues with any of those LCDs on an SKR 1.3.

AmiSMB, 2019-09-11:
I have had a similar issue to this but did some digging and found that my board was not booting due to having TMC_DEBUG enabled. It was this thread and the statement by reloxx13 that I thought I would try remarking it out to see if it fixed the problem. I think that there maybe a clash with EEPROM emulation as well as #15180 which allows the SD card to be seen via USB. I have opened up issue #15226 with my findings.

AbelCS, 2019-09-11:
I'm a bit lost with this issue, on SKR E3 DIP the other day all was working great with https://github.com/MarlinFirmware/Marlin/pull/15172 (TMC_DEBUG, EEPROM EMULATION, and all the stuff), today I pulled all new changes to my repo and with same configuration I've got blank screen, so I decided to revert to https://github.com/MarlinFirmware/Marlin/pull/15172 with the same config, and now I'm getting blank screen all time.

sjasonsmith, 2019-09-22:
I just rebased onto commit [7fc4f7b815949eb9ec4b8e7c5399d154d0f28c94](https://github.com/MarlinFirmware/Marlin/commit/7fc4f7b815949eb9ec4b8e7c5399d154d0f28c94) (from yesterday), and it seems to be working fine.

I'm on an SKR-mini-E3. Display comes up, and I'm able to query TMC_DEBUG info as well.

thisiskeithb, 2019-09-22:
> I'm on an SKR-mini-E3. Display comes up, and I'm able to query TMC_DEBUG info as well.

With emulated `EEPROM` enabled?

matheusmbar, 2019-09-22:
I've just setup Marlin firmware in my Ender 3 with SKR E3 DIP and the CR10 Stock display is working too. 

I'm at commit 7fc4f7b815949eb9ec4b8e7c5399d154d0f28c94

I have TMC_DEBUG, EEPROM_SETTINGS and EEPROM_AUTO_INIT enabled.

sjasonsmith, 2019-09-22:
> With emulated EEPROM enabled?

I have EEPROM_SETTINGS enabled, and the pins file defines FLASH_EEPROM_EMULATION. I confirmed I can save settings and they persist across reboots without an SD card installed.

boelle, 2019-10-12:
@vincentbaeten does it work for you with latest 2.0.x ?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle