MarlinFirmware/Marlin #15142: operator '==' has no right operand

3Dangel, 2019-09-02:
hello to all,
Just downloaded the newer version of Marlin 2xxbugs branch
made all the changes for my mks Robin board.
I do have the same settings on my other version that I downloaded from a month or so ago and it complies with no issue.
but now I'm getting the following error:

c:\marlinrobin\marlin\src\core\boards.h:321:72: error: operator '==' has no right operand
 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)
                                                                        ^
c:\marlinrobin\marlin\src\pins\pins.h:457:7: note: in expansion of macro 'MB'
 #elif MB(MKS_ROBIN)
       ^~

not sure if this is related to the recent changes on board's name (I recall seeing some update in regard to this, not 100% sure), how would this "==" logic symbol be rejected?
would it be possible to get some help?
thanks in advance

reloxx13, 2019-09-02:
post ur config pls

3Dangel, 2019-09-02:
wow man you are so fast!!!
I found my issue..... dummy newbbie error....

I was too quick on delete....
deleted "MOTHERBOARD"  from #define

closed 2019-09-02 by 3Dangel

reloxx13, 2019-09-02:
did u use ArduinoIDE or Visual Studio Code?

If u use VSC we could (if its not in already) check if motherboard is defined before compiling.

3Dangel, 2019-09-02:
using AtomPlatfomio....
now compiling,..........:-)