MarlinFirmware/Marlin #15143: [BUG] (MKS Robin Lite 1.1 - Won't boot anymore)

ETE-Design, 2019-09-02:
### Description

After updating to newest Bugfix 2.x the MKS Robin Lite 1.1 freeze at Updating 100% - Before 14 days ago it just worked...

### Steps to Reproduce

1. Build MKSLite.bin
2. Put SD Card in Printer]
3. Turn on and wait
4. Freeze at "Updating 100%"

**Expected behavior:** [Flash 100% and then reboot]

**Actual behavior:** [Flash 100% and freeze]

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3567204/Configuration.zip)

RudyBenner, 2019-09-03:
Try this
[image]

RudyBenner, 2019-09-03:
Try this
[image]

RudyBenner, 2019-09-03:
Missing bootscreen.h file in your zip file?

anabos1758, 2019-09-04:
I have the same issue with ADMINLab Gantry S. The Marlin firmware code from Makerbase GitHub repository works(not up to date with offical Marlinfw repository).

tpruvot, 2019-09-05:
may be fixed by #15172 on the robin, please try and report

anabos1758, 2019-09-06:
Thanks, The problem is fixed. It is working.

ETE-Design, 2019-09-06:
Anabos1758 - Same here... But now it won't find any files in my SD Card, same for you? Or can you share your configuration files then?

anabos1758, 2019-09-06:
ETE-Degin
I am able to print from SD. However I have issues with 3DTouch (BLTouch
clone) deploys during print. I have abort at this point.
Here is my Config. I believe I took yours and added my settings.   I am
experimenting with 3DTouch on ADIMLab Gantry-s. I have enabled few configs
for debug/experiment.

anabos1758, 2019-09-06:
[myconf.zip](https://github.com/MarlinFirmware/Marlin/files/3584387/myconf.zip)
My Config files

RudyBenner, 2019-09-06:
I hope PC11 is not defined elsewhere.....

closed 2019-09-06 by ETE-Design

anabos1758, 2019-09-06:
Thanks for the suggestion. I suspect electrical (noise or poor connection)
or faulty sensor (may need probe adjustment). However I added the following
to my code.
//Using WIFI module's IO Pin for servo
#ifdef NUM_SERVOS
 #ifdef PC11
  #undef PC11
 #endif
#define SERVO0_PIN PC11 //WIFI IO2 PIN
#endif

tpruvot, 2019-09-08:
its not good to undef PC11, its SERVO0_PIN you "could" undefine, but i dont think its preset