MarlinFirmware/Marlin #15145: [BUG] Filament change freezes printer (M600)

albert2004, 2019-09-03:
# Description
Printer freezez after changing filament

### Steps to Reproduce

1. Issue M600 to change filament.
2. Wait until extruder is parked and unload procedure finishes
3. Replace filament.
4. Press button to confirm finish or replacement.

After filament change and confirmation by pressing button, on serial console there is counting down from 8 to 2 (seconds).
At last point, I hear short extruder motor activation and printer freezes. LCD is blank, no reaction for serial commands.

**Expected behavior:** print resumes

**Actual behavior:** freeze of printer, needed cold restart

#### Additional Information
GIT revision: 12c595c61f4b3e0be2baa6a906718e155a797ea3

thisiskeithb, 2019-09-03:
This probably related to https://github.com/MarlinFirmware/Marlin/issues/15138 which has been patched, so download a fresh copy and see if the issue persists.

albert2004, 2019-09-07:
Yes, I can confirm. New version works again.
Closing.

closed 2019-09-07 by albert2004