MarlinFirmware/Marlin #15147: [BUG] Stealthchop movement inability

MarioStoilov, 2019-09-03:
I recently flashed marlin 2.0 from 2.09.2019 and am having strange behavior on TMC drivers. I have TMC2130s on all motors, except for Y (where it is a TMC2209). Operating in spreadcycle seems to give me no problems, but having the drivers in stealthchop gives huge problems.

A short video to illustrate the problem: https://www.dropbox.com/s/z20pl6q7abyvy0u/2019-09-02%2021.01.38.mp4?dl=0

The issue is present on all axis using stealthchop (even on the 2209). I tried changing current values, disabling hybrid threshold, using junction deviation ans s curve. It seems it has something to do with speed, as shorter moves do not reproduce this problem .In the video, I'm doing 50mm moves via repetier-host, doing 10mm moves is ok (and doing whatever move is ok in spreadcycle).

I currently am out of Ideas on what should I check, looking for some guidance. I'm attaching configs. Any help would be appreciated. 
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3568487/Archive.zip)

gloomyandy, 2019-09-03:
Your config shows you using 256 microsteps I think Marlin will struggle with ensuring the correct timing for this at higher speeds (which fits with what you are seeing as longer moves give time for the speed to increase). If the pulses generated are not "smooth" this seems to cause problems for Stealthchop. I would suggest that you use 16 (or 32)  microsteps and set the driver to interpolate to 256 (which will typically give smoother movement). I doubt if you will see any sort of reduction in print quality. If you do change this you will also need to change the steps/mm setting to match.

You may also want to try setting: 
#define SQUARE_WAVE_STEPPING 
and/or:
 #define MINIMUM_STEPPER_PULSE 2
Both of which allow Marlin to produce a cleaner set of control pulses for the steppers.

MarioStoilov, 2019-09-03:
So far, this seems to work. Wondering if there is a way to get 256 steps working on speeds like 80mm/s (which is not even that fast)

gloomyandy, 2019-09-03:
You may want to do some searching on using 256 microsteps vs using interpolation. The general consensus seems to be that there is very little (if any) improvement in print quality over using a lower microstep value and interpolation (at least with typical 3D printing hardware). Using lower values certainly puts much less load on the mcu. The timing of the interpolated steps is generated on the driver chip and will almost certainly be better than the timing of the steps generated under software control (even on a 32bit mcu). Arguably this will provide smoother movement. Some folks have reported being able to identify improvements when using 32 or 64 microsteps (typically still with interpolation to 256), so you may want to investigate those.

closed 2019-09-03 by MarioStoilov