MarlinFirmware/Marlin #15148: [BUG] LCD Bed Levelling doesn't work when extruders set to 2

jewjudge90, 2019-09-03:
Apologies in advance if this is the way it is supposed to work but after adding a 2nd hot end to my 3d printer my current buid of Marlin 2.0 (2019-07-29) I can no longer use the LCD bed levelling feature when Extruders is set to 2 in the firmware. 

Selecting level bed just returns immediatly to the home screen while the printer homes itself, no "waiting for homing" or any messages on the screen. 

Having enabled all the serial debugging commands after the printer and monitoring the serial port, selecting Level bed just returns an "OK" after the printer has returned to the main screen and the homing has finished.

If I make only 2 changes to the firmware ( the Number of extruders to 1 from 2 and the corrosponding Temp sensor 1 value to 0 (wont compile with extruder>1 and tempsensor 1 enabled)) the LCD bed levelling option works again.

Did do some reserach on the forums but not found any answers and there was nothign in the code to say that it shouldn't work with a dual hotend setup so perhaps it is a bug?

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3569608/config.zip)

shitcreek, 2019-09-05:
Have you tried testing using one of the dummy sensors: 999, 998?
Not sure if it matters but you don't have your hotend offsets defined:
> //#define HOTEND_OFFSET_X { 0.0, 0.00 } // (mm) relative X-offset for each nozzle
> //#define HOTEND_OFFSET_Y { 0.0, 5.00 }  // (mm) relative Y-offset for each nozzle
> //#define HOTEND_OFFSET_Z { 0.0, 0.00 }  // (mm) relative Z-offset for each nozzle

boelle, 2019-09-24:
@jewjudge90 still having problems?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle