MarlinFirmware/Marlin #15149: [BUG] usage of UART0 and USB problem : reset/kill

kakou-fr, 2019-09-03:
# Description

I am trying to build an MMU2 clone with an SKR mini board. I try to connect the MMUv2 to a
SKR 1.3, so i use the TFT board on the two board. The firmware of this MMUv2 was tested before on a gt2560 (using the HardwareSerial on the GT2560) without issue.

The skr v1.3 is also connected on octoprint with USB.

1) Power the MMUv2 and next power the printer : boot loop (just the Marlin logo)
2) Power the printer, power the mmu, and next connect to octoprint : the board reboot and boot loop
3) Power the printer, connect to octoprint, next power the MMUv2 : ok.
3.1) if I use the MMUv2 from the Marlin screen menu, all is ok
3.2) if I send any gcode from the octoprint interface (so via the USB serial), the printer reset (and boot loop)

SO it looks like a conflict when using the USB and UART0 at the same time.

shitcreek, 2019-09-03:
Firstly, please follow the template and provide a zip of your config and more information.

Secondly, this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

kakou-fr, 2019-09-03:
For me it's a bug in the PRUSA_MMU par code, when i desactivate it, Marlin boot correctly. 
So maybe an issue in the feature/prusa_mmu serial code, watchdog or ...

kakou-fr, 2019-09-03:
ok I find the problem in the mmu2.cpp code, if I increase the size of this buffer from 16 to (2048 (big value for testing), all works correctly  : 

char MMU2::rx_buffer[16], MMU2::tx_buffer[16];

shitcreek, 2019-09-03:
That's good.
I'm tagging this as a potential bug for now and see if anyone that's more familiar with this would chime in.

kakou-fr, 2019-09-03:
thks, i try to correct it, maybe i found the issue, but i need to test. i will rename the title because it's an MMU specific bug

kakou-fr, 2019-09-11:
Hi, what is the best function to force a E move in  ? I tried to extend the MMUv2 feature for an open-source firmware, and I need to force a E move in order to insert the filament in the extruder while the mum push it

shitcreek, 2019-09-11:
Use M83 or G91 to use relative mode, then G1 Exxx (xxx = amount).

kakou-fr, 2019-09-11:
no a code, but the name of a function in the Marlin source code :)

boelle, 2019-09-24:
@kakou-fr still having issues?

kakou-fr, 2019-09-25:
the issue is always present in the code.
if the mmu send too many octet to marlin when it start (on the serial link), marlin crash (the RX size is not correctly managed).
I patched it by increasing the size of the RX/TX buffer in the mmu code, but it's not the correct way :)
I will publish a better update with some new feature in the mmu part ASAP

saeugetier, 2019-10-18:
I'm having the same problems with my SKR Mini E3. Could you please send a pull request? Or should I prepare one?

kakou-fr, 2019-10-19:
you can do it :) i prepare another patch in order to include 
- include load/unload sequence in Marlin
- wait the acknowlegement during C0 (and not start the print if the filament in not inserted for mk3s-mmu sensor on the bondtech gear)
- add a eject length after the unload/ramming to be sure that the filament is not blocked in the gear

boelle, 2019-10-21:
@saeugetier made a PR yet?

saeugetier, 2019-10-23:
@boelle haven't done it yet. I want to test the change a bit and then I do the pull request.

closed 2019-10-24 by thinkyhead

BastR, 2020-03-25:
> you can do it :) i prepare another patch in order to include
>
> * include load/unload sequence in Marlin
> * wait the acknowlegement during C0 (and not start the print if the filament in not inserted [...]

@kakou-fr It's very, very interesting, were you able to make any progress on this issue during the lockdown?

kakou-fr, 2020-03-25:
@BastR yes, but wih the klipper version.
I use it since 3 month, and all the patch are now integrated in klipper (and the DIY MMU2S config file).

For Marlin, I will publish all my last patch here : https://github.com/kakou-fr/Marlin/commits/bugfix-2.0.x
This last version was working for me with all the feature listed above

BastR, 2020-03-25:
@kakou-fr great, because even though the MMU2 works with Marlin, there are a lot of bugs that I hope will be fixed with your future PR.

I'm going to try to incorporate your changes into my current marlin to test.

Merci et bon courage