MarlinFirmware/Marlin #15152: Error found MKS LCD 12864a

BucaneroAlmeria, 2019-09-03:
I have this error when putting the MKS LCD 12864a in Marlin "Marlin \ src \ lcd \ dogm \ ../../ inc /../ pins / lpc1768 / pins_MKS_SGEN_L.h: 247: 28: error: 'P1_0' was not declared in this scope "
[https://ae01.alicdn.com/kf/U5ec07180ced244c1a02dd9a84b09f173g.png](url)
I found the error inside pins_MKS_SGEN_L.h
#if ENABLED (MKS_MINI_12864)
#define DOGLCD_CS P0_17
**#define DOGLCD_A0 P1_0**
needs to be changed to **#define DOGLCD_A0 P1_00** There is a 0 missing.

thisiskeithb, 2019-09-03:
See PR https://github.com/MarlinFirmware/Marlin/pull/15153.

RudyBenner, 2019-09-03:
Same error in several spots.

thisiskeithb, 2019-09-03:
> Same error in several spots.

Did you review the PR @RudyBenner?

BucaneroAlmeria, 2019-09-03:
I do not understand your last message, it is the first time I do this and I do not know how it works, I am also Spanish and I use the Google translator, I do not understand well, so I do not know what you want to tell me.

thisiskeithb, 2019-09-03:
> I do not understand your last message, it is the first time I do this and I do not know how it works, I am also Spanish and I use the Google translator, I do not understand well, so I do not know what you want to tell me.

This will be fixed soon.

thisiskeithb, 2019-09-04:
The fix has been merged and this issue can be closed.

closed 2019-09-04 by BucaneroAlmeria