MarlinFirmware/Marlin #15155: [BUG] (short description)

zombu2, 2019-09-04:
### Description

Reprapdiscount GLCD displays nothing but rotary and beeper and sd card work on bigtreetech SKR PRO board 
Tried on other mcu's like LP based and display works correctly issue only present on stm mcu

### Steps to Reproduce

Build latest marlin for SKR PRO and use reprapdiscount glcd

**Expected behavior:** [What you expect to happen]
display info on screen
**Actual behavior:** [What actually happens]
nothing

#### Additional Information
The link below is the tree it stopped working 
https://github.com/MarlinFirmware/Marlin/tree/3ac76f048973e1ecb5ea6f2c50eb5990bbd6a5b9

as for config i used a stock config and enabled reprapdiscount glcd thats it 
it works with a commit from before above link and at that particular link it stopped 

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-09-04:
This is related to https://github.com/MarlinFirmware/Marlin/issues/15141

zombu2, 2019-09-04:
it's related to these files 
if i drop in older ones the display starts working on newest pull 

ultralcd_st7920_u8glib_rrd_AVR.cpp
ultralcd_st7920_u8glib_rrd_AVR.h
u8g_dev_st7920_128x64_HAL.cpp

zombu2, 2019-09-07:
fixed

closed 2019-09-07 by zombu2