MarlinFirmware/Marlin #15158: BLTOUCH connetcted to user defined Z_MIN_PROBE_PIN is not working

pjanco, 2019-09-04:
Hi,
I have Ender 3 pro printer and I want to use BLTOUCH and filament runout sensor at the same time.

My idea is to use BUZZER_PIN 27 as output for controlling the BLTOUCH.
Expansion pin 29 (hidden under the capacitor) will be probe sense input.
Unused original Z end stop plug on pin 20 will be used for filament runout sensor.

I have disabled alot of features in my project to reduce code size. Buf I am pretty sure that this has nothink to do with my problems.

I am almost sure my config file and hardware connection are OK. If I navigate to LCD BLTOUCH menu, BLTOUCH is moving up and down. If I start homing, sensing stick does not move down, Z motor keeps moving down and nozzle crash into a bed.

I think I found 3 bugs related to BLTOUCH and fix them. It help, but it is still not fully functional!

**Founded bugs:**
In file Conditionals_LCD.h:
`#define HAS_BED_PROBE (ENABLED(FIX_MOUNTED_PROBE) || ENABLED(Z_PROBE_ALLEN_KEY) || HAS_Z_SERVO_PROBE || ENABLED(Z_PROBE_SLED) || ENABLED(SOLENOID_PROBE))`
My fix:
`#define HAS_BED_PROBE (ENABLED(FIX_MOUNTED_PROBE) || ENABLED(Z_PROBE_ALLEN_KEY) || HAS_Z_SERVO_PROBE || ENABLED(Z_PROBE_SLED) || ENABLED(SOLENOID_PROBE) || ENABLED(BLTOUCH))`

`#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))`
My fix:
`#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && (ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) || ENABLED(Z_MIN_PROBE_ENDSTOP)))`

In file SanityCheck.h:
`#if Z_HOME_DIR < 0 && DISABLED(USE_ZMIN_PLUG)`
My fix:
`#if Z_HOME_DIR < 0 && DISABLED(USE_ZMIN_PLUG) && DISABLED(Z_MIN_PROBE_ENDSTOP)`

**Current status after this bug fixies is:**
If have defined #define USE_ZMIN_PLUG, the BLTOUCH is working fine.

If I have commented //#define USE_ZMIN_PLUG (because I need this plug for filament sensor) the BLTOUCH is not working anymore.

In attachment is my whole project:
[Marlin-1.1.x-pjanco.zip](https://github.com/MarlinFirmware/Marlin/files/3573623/Marlin-1.1.x-pjanco.zip)

pjanco, 2019-09-04:
I just solve this problem:

My changes in file Conditionals_LCD.h

```
//#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && DISABLED(USE_ZMIN_PLUG))
```

In file SanityCheck.h:

```
//#if Z_HOME_DIR < 0 && DISABLED(USE_ZMIN_PLUG)
#if Z_HOME_DIR < 0 && DISABLED(USE_ZMIN_PLUG) && (!HOMING_Z_WITH_PROBE)
```

In file Pins.h

```
#if DISABLED(USE_ZMIN_PLUG)
  #undef Z_MIN_PIN
  //#define Z_MIN_PIN          -1
  #if ENABLED(Z_MIN_PROBE_ENDSTOP)
    #define Z_MIN_PIN Z_MIN_PROBE_PIN
  #else
    #define Z_MIN_PIN -1
  #endif
#endif
```

But I think better solution for me will be customizing file pins_MELZI_CREALITY.h and redefining default Z_STOP_PIN from 27 to 29. Then I can use Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN insted of Z_MIN_PROBE_ENDSTOP.

Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN configuration is used by everyone and it is better tested.

I forgot to mention that I am referring to Bugfix-1.1. I did not try 2.0, but According to mentioned header files in Bugfix-2.0 there will be the exact same problem.

boelle, 2019-09-24:
@pjanco is this still a problem?

pjanco, 2019-09-24:
> @pjanco is this still a problem?

Not for me anymore. I have changed header files in my project a compiled my firmware. But maybe header files in original branch should be changed in this way too. So anyone can configure Z_MIN pin to alternative function.

closed 2019-10-12 by boelle

diogosalazar, 2019-10-13:
Sorry, I notice this was closed but wanted to chime in and say that I am in a similar situation.
Have a BLTouch on a Creality Melzi board utilizing pin 7 as opposed to the Z min endstop plug.

Had to manually set HOMING_Z_WITH_PROBE to true in order to be able to home the Z axis with the probe on an alternate pin when the original machine endstop had been removed.

Shouldn't the HOMING_Z_WITH_PROBE be redefine to account for this use case? Tks

Roxy-3D, 2019-10-13:
> Shouldn't the HOMING_Z_WITH_PROBE be redefine to account for this use case? Tks

It is my opinion that the firmware should be re-organized to just let the user say "The Z Probe is on pin x" and everything else just works out right.

The whole Z-Probe configuration process for a new machine is confusing.

pjanco, 2019-10-13:
My final solution is to customize file "pins_MELZI_CREALITY.h" instead of "Configuration.h"
```
#undef BEEPER_PIN
#undef Z_STOP_PIN
#undef SERVO0_PIN
#define BEEPER_PIN     	   -1   
#define Z_STOP_PIN 	   29
#define SERVO0_PIN 	   27 
```

With this, I can use default configuration `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` in "Configuration.h".

And you also have to disable BEEPER_PIN in case you are using buzzer signal from LCD cable as Bltouch PWM. Most people do no care about this and then BLtouch can have random behavior.

`#define` inside "pins_MELZI_CREALITY.h" have bigger priority during compilation then inside "Configuration.h".

**And about pin 7:**
A tried same configuration at the very begin because it is proposed there:
https://www.reddit.com/r/ender3/comments/c02q88/the_definitive_guide_to_wiring_bltouch_v3_on_icsp/

And it was mistake because ISP pins are shared with SDCARD. If you will redefine ISP pins for BLtouch then SDCARD will not be functional anymore.