MarlinFirmware/Marlin #15159: [FR] (BUZZER deactivation feature when using BLTOUCH on pin 27)

pjanco, 2019-09-04:
Hi,
I have ender 3 and BLTOUCH connected to pin 27 - BUZZER_PIN. I read couple of tutorials from internet and follow instructions.

It is working, but sometimes sensing stick ramdomly moves down with no reason.
I believe it is caused by buzzer functionality in firmware. When firmware try to beep on buzzer, it interfere with BLTOUCH connected to pin 27. 

For fix this, It have commented line `//#define SPEAKER` in file Configuration.h,
and deactivated BUZZER in pins_MELZI_CREALITY.h:
```
#undef BEEPER_PIN
#define BEEPER_PIN -1
``` 

All of the tutorials on the internet does not care about deactivation of buzzer pin in firmware.
I think developers should preparae some macro for deactivating buzzer if this pin is used for BLTOUCH.

thisiskeithb, 2019-09-04:
This is already handled in the pins file:

https://github.com/MarlinFirmware/Marlin/blob/d9bbe0ff02530239dfa726b87e8303ff427755ba/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L56-L58

You can't use `BLTOUCH` without first disabling `SPEAKER`. With `SPEAKER` disabled, there are no buzzer/speaker functions.

pjanco, 2019-09-04:
Hi,
I am using Bugfix-1.1 and section you mentioned is missing here.
You are referring to branch 2.0.

I already have SPEAKER undefined in my 1.1 configuration because of size reduction and I do not agree with you.
```
//
// SPEAKER/BUZZER
//
// If you have a speaker that can produce tones, enable it here.
// By default Marlin assumes you have a buzzer with a fixed frequency.
//
#define SPEAKER
```

According to this comment in Configuration.h, if SPEAKER is undefined, BEEPER_PIN is still used as simple ON/OFF output.

If I have SPEAKER undefined and BEEPER_PIN defined to 27, my code size is 129624.
If I have SPEAKER undefined and BEEPER_PIN defined to -1, my code size is 128462.

thisiskeithb, 2019-09-04:
`Marlin 1.1.x` is no longer being updated and all development is focused on `bugfix-2.0.x`.

Please try `bugfix-2.0.x` and see if your problem persists.

AnHardt, 2019-09-04:
That look like an error, in the pins  file and its understanding.

If the BEEPER_PIN is used by the sound generating code or not is only depending on its existence. To disable sound generation eiter undefine BEEPER_PIN or set it to -1 or comment all its definitions.
SPEAKER makes the difference it the pin is only switched on when the sound starts and off when it ends and (with SPEAKER) is toggled during the on time with the tone frequency - ideally with a timer.

So if 
```
 #if DISABLED(SPEAKER) && ENABLED(BLTOUCH) 
   #define SERVO0_PIN 27 
 #endif 
```
makes any sense, because of the timer, not because of the pin. But i guess this lines are not expressing the intention of the author.
@InsanityAutomation ?

InsanityAutomation, 2019-09-04:
IIRC original intent was only for bltouch enabled to set the pin, but an undef of the beeper pin there may be prudent as well since they may overlap. With the speaker a pwm output matching a bltouch command is more likely to be generated. Would need to get just the right output with a beeper alone... It should ignore anything out of range, such as full on / full off.

This may go back to the inaccurate PWM outputs Bob the Khun measured on the melzi boards awhile ago.

Regardless, some of the beeper / buzzer stuff needs tweaking as it currently doesnt work for ExtUI to trigger play tone if there is no physical pin. ill make it around to that soonish....

sjasonsmith, 2020-07-18:
It looks like this was already done for `MELZI_CREALITY`:

```
#if ENABLED(BLTOUCH)
  #define SERVO0_PIN                         27
  #undef BEEPER_PIN
#endif
```

I'm closing this, presuming it is resolved.

closed 2020-07-18 by sjasonsmith