MarlinFirmware/Marlin #15160: [FR] Allow G29 to adjust Z_MAX_POS and Z_MIN_POS

marcio-ao, 2019-09-04:
One frequent complaint we get from our customers is that running G29 at the start of every print is very slow. Even though Marlin has the ability to store bed leveling information, in practice we have found that doing a print without G29 leads to printing in the air.

I suspect what is going on is that G29 does two things. 1) it measures points on the bed and determines the **orientation** of the level plane 2) it measures the **bed height** and re-adjusts the Z=0 to coincide with the measurement (taking into account the Z_PROBE_OFFSET).

On a printer that homes to Z_MAX, the initial coordinate system after a power on set by G28 based on the value of Z_MAX_POS. Once the user runs G29, the probing may determine the bed to be at some position other than Z=0 and the coordinate system will be shifted to correct for this, but AFAIK this adjustment is not saved anywhere. This means that if the user reboots the printer and attempts a print without G29, the print will happen either in the air or below the bed surface.

The thing is that once G29 probes the bed, it has knowledge about the bed's Z position that likely is at odds with the coordinate system established by Z_MIN_POS and Z_MAX_POS. To ensure this information is preserved, Z_MIN_POS and Z_MAX_POS could be adjusted and saved in the EEPROM. 

@thinkyhead  What is above is my best understanding of what is happening, but I do not understand bed leveling sufficiently enough to know whether this explanation is correct.

@Roxy-3D: It is possible that your UBL code already accounts for this problem? For historical reasons, Lulzbot continues to use AUTO_BED_LEVELING_LINEAR, but if your new code accounted for this problem, it definitely would be a reason for us to switch.

golfromeo-fr, 2019-09-04:
I am just wondering if it is related to z probe offset I am using with babystepping that once could be saved but not much currently (I am using SKR Pro and I am not sure of eeprom (or emulation) is working properly on 2.0.x

//#define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping

Vertabreak, 2019-11-13:
sounds like you just need to switch to UBL bed leveling that is run then stored on the machine.

kpishere, 2021-02-14:
I'll add to this ticket with a more specific request about Z_MAX.

For the case of homing at Z_MAX_POS and then performing bed leveling with a z-probe, I'd ask that the Z_MAX_POS value is adjusted by the average value of the points measured.

Here is a couple of posts on the RepRap forum.  The thing is, in this case of homing at max and probing at min, the Z_MAX_POS value is critical in z-calibration height.

https://reprap.org/forum/read.php?415,881521,881523#msg-881523

p1xiedwarf, 2023-12-16:
hi, I have the same issue, I think, I am searching and searching. Loading a mesh will make the printer print in the air. I have a new delta printer with marlin 1.1 and I don't have the time and knowledge to flash new firmware with all the settings. I did a M503 to get all settings but it didn't return any relevant value for z max height. However, the z offset I try to save to eeprom when I do autolevel and measure z pos, is -15mm. I believe it never saves since the distance when printing after loading the mesh is approximately that. When I don't load the mesh it knows where my bed is. I read somewhere that the z offset can't have a negative value so I am planning to bring the xyz endstops 20mm down, so the z offset when loading the mesh has a positive value. It gets me crazy, I can't understand this behaviour.