MarlinFirmware/Marlin #15161: Marlin 2.0.x temp sensor 20 cause firmware to crash on powerup

dingbatca, 2019-09-04:
Steps to reproduce:
Current (9-4-2919) 2.0.X tree from github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip

Only changes made to stock Configuration.h
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200
  #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
#define TEMP_SENSOR_0 20
#define TEMP_SENSOR_BED 5
#define SDSUPPORT

If I compile with TEMP_SENSOR_0 5, the board power up and I get full communications.  If I set the TEMP_SENSOR_0 to 20, I get nothing from the serial console and  storage does not work.

If i remove half the raw temp values in MarlinFirmware\Marlin\src\module\thermistor\thermistor_20.h it works.

No compile errors, everything seems to build clean.  I am kind stumped on this one

AnHardt, 2019-09-04:
Witch half? (First? Second? Every second value? Random?) 5V or 3.3V on amplifier? Voltage divider? Any hints on the display? Reboots?

The likely scenarios are:
Connected to a *normal* thermistor input with 4.7K pullup ADC produces much to high values, reads above MAXTEMP and fails for that before the USB serial is ready.
Amplifier on 5V, without voltage divider connected to 3.3 V ADC-pin. ADC produces 5/3.3 to large values (and is destroyed above ~500°C), is above MAXTEMP right at the boot and fails with MAXTEMP before the USB-serial is ready.
Or a combination of these.

dingbatca, 2019-09-04:
Nothing connected at this point (open circuit).  On only tried deleting the secound half, assuming the array was to big.  Never though about the values there in.

I don't have a display.  I do everything through octoprint.

Great call on the MAXTEMP.  
//#define HEATER_0_MAXTEMP 275, and now I see "T:1100.00 /0.00 B:21.00 /0.00 @:0 B@:0" WHELP!  There is the problem.  We can close this bug.

I am brand new to Marlin, moving over from Smoothieware.  So please bare with me.

My setup has a E3D PT100 amplifier connected to my SKR V1.3 on pin 1.30.  How do I point my "TEMP_SENSOR_0" to pin 1.30?  The LPC1768 CPU is 3.3V.  I am planning on using this adjusted temp scale for 3.3V:
https://e3d-online.dozuki.com/Answers/View/260/How+does+one+use+the+PT100+Amp+Board+with+a+3.3v+controller,+eg+RADDs

Current all the temperature stuff work perfectly in Smoothieware.

AnHardt, 2019-09-04:
This is not a beginners forum but a bug-tracker. Please close this issue.
While testing and not having connected the sensors, try the fake-sensor-types 999 and/or 998.

closed 2019-09-04 by dingbatca

Powerprobot, 2019-09-05:
...
> My setup has a E3D PT100 amplifier connected to my SKR V1.3 on pin 1.30. How do I point my "TEMP_SENSOR_0" to pin 1.30? The LPC1768 CPU is 3.3V. I am planning on using this adjusted temp scale for [...]
...
Nonsense... Try Sensor 1, it works / fit perfectly. So many E3D reworks with SKR1.3 in the Internet can't lie.