MarlinFirmware/Marlin #15163: Error on M122 command

Schullee, 2019-09-04:
Hello,

I have a problem with my board. I want to operate the board with UART, have installed on X Y Z (Dual Z) TMC2208 driver. As soon as I enter the command M122 in S3d, I get the following feedback:

READ: Driver registers:
READ: X 0xC0:0C:00:00
READ: Y 0xC0:0C:00:00
READ: Z 0x00:00:00:00 Bad response!
to read: 
to read: 
READ: Testing X connection... Ok
READ: Testing Y connection... Ok
READ: Testing Z connection... Error: All LOW

In Marlin, the drivers are set to TMC2208. Drivers have already been replaced to rule out a defective driver. Can you help me?

Grogyan, 2019-09-04:
This is not a bug, 
UART PDN soldered, and board must not be powered from USB

Schullee, 2019-09-04:
The drivers are abble for UART without Soldering.

Grogyan, 2019-09-04:
Best ask in the Marlin Discord

thinkyhead, 2019-09-04:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-09-04 by thinkyhead