MarlinFirmware/Marlin #15167: z_min endstop not stopping dual z steppers in autohome

michaelduffy123, 2019-09-05:
when autohoming z axis with dual stepper drivers, Tmc2209, even when dual end stop commented out, z min endstop only triggers one zmotor to stop moving on  skr1.3

GMagician, 2019-09-05:
not clear what you mean with 

> even when dual end stop commented out

but dual stepper requires dual endstops

shitcreek, 2019-09-05:
Please follow the template and provide more information. Also, have you exhausted other sources and find that this is indeed a bug?

For best results getting help with configuration and troubleshooting, please use the following resources:

MarlinFW.org Marlin Documentation
RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

If you are using the spare extruder stepper, then you must enable:
`Z_DUAL_STEPPER_DRIVERS` in configuration_adv.h
and `Z2_DRIVER_TYPE` enabled, **not** `E1_DRIVER_TYPE`

@GMagician - dual stepper **does not** require dual endstops.

boelle, 2019-09-06:
@shitcreek it has been suggested on discord that we get a bot to precheck that issues are created like they should be, just like octoprint does it. maybe poke your "boss" and ask if its not time to do some testing on it

closed 2019-09-07 by michaelduffy123