MarlinFirmware/Marlin #15168: [BUG] Z_SAFE_HOMING don´t work properly when MANUAL_X_HOME_POSITION is enabled

Parkrcz, 2019-09-05:
Hello there,

recently I moved from Bowden to direct drive extruder and due to this change, I had to change X_ENDSTOP position from MIN to MAX also. I modified all necessary stuff related to new endstop position in Marlin firmware (Z_SAFE_HOMING enabled all the time) and then I tried Auto_Home my printer. 
 Homing X and Y-axis were fine, but instead of moving hot end to the Z_SAFE_POSITION (in mid of bed) for homing Z-axis, it hits X_MAX_ENDSTOP again and started homing outside of the print bed. 
 I've checked x_stepper_motor direction (was correct), decreased feedrates for homing (all are below threshold values), increased HOME_BUMP value (to be sure X_ENDSTOP is not triggered after homing) and I also modified MANUAL_X_POSITION (was all the time enabled) => all without success, hot end still moving to the right.

  Then, I tried to disable MANUAL_X_HOME_POSITION (just for fun, because this feature works well with X_MIN_ENDSTOP) and guess what => Z_SAFE_HOMING moved hot end to the center of the bed (desired position).

I know that bed can be centered also via modifying "Travel limits (mm) after homing, ...", but still this issue can cause hard times to anybody else.

Thank You in advance and I apologize for my grammar.
-->

### Description

Z_SAFE_HOMING feature doesn´t work preperly when X_MAX_ENDSTOP is enabled  = > after X/Y_Axis homing(works fine) hot end moves only to the right, hits X_MAX_ENDSTOP again and start Z-axis homing outside of the bed.

### Steps to Reproduce

1. Use X_MAX_ENDSTOP
2. Use MANUAL_X_HOME_POSITION
3. Enable Z_SAFE_HOMING

**Expected behavior:** After X/Y_axis homing hot end moves to the Z_SAFE_POSITION and start homing Z_Axis.

**Actual behavior:** After X/Y_axis homing hot end moves only to the right, hits X_MAX_ENDSTOP again and perform Z_Axis homing outside of the bed.

#### Additional Information

* Cofiguration.h + Configuration_adv.h will be added later.

boelle, 2019-09-24:
@Parkrcz is this still a problem?

Parkrcz, 2019-09-25:
I was not able to solve this, so I moved back to the original configuration => X_MIN_ENDSTOP instead of the X_MAX_ENDSTOP, since that, no issue.

boelle, 2019-10-21:
will close this one as its seems fixed and that there has been no activity

we can always reopen if this is still an issue and it can be reproduced easy

closed 2019-10-21 by boelle

thinkyhead, 2019-10-21:
The homing direction —and min/max endstop— is set with the `X_HOME_DIR`, `Y_HOME_DIR`, and `Z_HOME_DIR` settings, and not by settings of the form "`X_MAX_ENDSTOP`".