MarlinFirmware/Marlin #15170: [BUG] Marlin 2.0 max endstop inverting

Sjoerrdd, 2019-09-05:
### Description
I'm trying to install Marlin 2.0 on my Ender 5.
I'm using the SKR V1.3 with TMC2209 stepper drivers.
I followed TeachingTech's video about that, but since I've an Ender 5 I've to use max_endstops instead of min_endstops.

When I set `X_MAX_ENDSTOP_INVERTING` and `Y_MAX_ENDSTOP` to "true", I get the following error: 

```
 #error "SENSORLESS_HOMING requires X_MAX_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMAX when homing to X_MAX."
```

**Expected behavior:** [What you expect to happen]
It should upload to my printer and work.

**Actual behavior:** [What actually happens]
It doesn't upload the firmware and I get errors.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3580270/Configuration.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-09-05:
You only have to set X_MAX_ENDSTOP_INVERTING to true if your endstop is NO - normally open. 
This might be relate to #14464

Sjoerrdd, 2019-09-05:
Thanks @shitcreek,
I get the following error: "SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMIN when homing to X_MIN."

Can't find a solution.

shitcreek, 2019-09-05:
If you want to go to X max, your X homing direction should be 1 and you should use the x max endstop plug.

Sjoerrdd, 2019-09-06:
@shitcreek I don't use the endstop plugs, because I want to home it sensorless.

shitcreek, 2019-09-06:
For Sensorless Homing, you still need to declare the correct endstop plug. With `ENDSTOPPULLUPS` enabled, Marlin checks to see which endstop plug is used and activates the corresponding `ENDSTOPPULLUP` (x min, x max, y min,.. etc.).

To sensorless home to X max, the following must be set:
`#define USE_XMAX_PLUG`
`#define X_MAX_ENDSTOP_INVERTING false` - I don't think it matters whether it's set to true or false, long as `X_MAX_ENDSTOP_INVERTING` is defined.
`#define X_HOME_DIR 1`
&
` #define SENSORLESS_HOMING`

boelle, 2019-09-24:
@Sjoerrdd is this still a problem?

Sjoerrdd, 2019-09-24:
I'll check it later this week :)

---- On di, 24 sep 2019 21:10:31 +0200 notifications@github.com wrote ----

@Sjoerrdd is this still a problem?

Sjoerrdd, 2019-09-26:
Fixed!

boelle, 2019-09-26:
closing then

closed 2019-09-26 by boelle