MarlinFirmware/Marlin #15173: [BUG] incorrect 'ok' response after M501

alexlinno, 2019-09-05:
### Description

marlin responses with 'echo:ok' after M501 instead of 'ok' this is not accepted by octoprint

### Steps to Reproduce
use marlin bugfix 2.0 of 4/9/2019 SHA-1: 6b05d5d65d0b4f557021c0ed5c47ec4e5e97b43c
connect to marlin
any command gives a 'ok' response
send a M501 this is ended with 'echo:ok' and after that all responses/acks use 'echo:ok' instead of 'ok' 
example:
Recv: ok
Send: G1 Z10 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: M501
Recv: echo:V69 stored settings retrieved (604 bytes; crc 23713)
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo: M92 X200.00 Y200.00 Z800.00 E837.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X400.00 Y400.00 Z8.00 E50.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X600.00 Y600.00 Z100.00 E5000.00
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P400.00 R1000.00 T400.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 W I0 J0 Z0.45250
Recv: echo:  G29 W I1 J0 Z0.10500
Recv: echo:  G29 W I2 J0 Z0.14000
Recv: echo:  G29 W I0 J1 Z0.02500
Recv: echo:  G29 W I1 J1 Z-0.01000
Recv: echo:  G29 W I2 J1 Z-0.03750
Recv: echo:  G29 W I0 J2 Z0.15000
Recv: echo:  G29 W I1 J2 Z0.08000
Recv: echo:  G29 W I2 J2 Z-0.25000
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B70 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P14.03 I0.65 D75.57
Recv: echo:  M304 P104.09 I12.06 D598.74
Recv: echo:Retract: S<length> F<units/m> Z<lift>
Recv: echo:  M207 S1.50 W13.00 F1800.00 Z0.00
Recv: echo:Recover: S<length> F<units/m>
Recv: echo:  M208 S0.00 W0.00 F480.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.10
Recv: echo:Stepper driver current:
Recv: echo:  M906 X800 Y800 Z800
Recv: echo:  M906 I1 Z800
Recv: echo:  M906 T0 E1200
Recv: 
Recv: echo:Driver stepping mode:
Recv: echo:ok

It seems to have happend when going from EEPROM v68 to v69

**Expected behavior:** [What you expect to happen]
always a 'ok' response instead of 'echo:ok'
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/3580992/Marlin_config.zip)

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ManuelMcLure, 2019-09-05:
Please try the following: Enable StealthChop on your drivers with ``M569 X Y S1`` and see if you have the same problem. My guess is that you won't see it, and this is specific to the "Driver stepping mode:" not actually printing anything.

alexlinno, 2019-09-06:
I have StealthChop enabled by default..but you mean..it is not save in the EEPROM since it is set in FW.....ok maybe..so probably doesn't have to be stealthchop ..as long as it is set to something

alexlinno, 2019-09-06:
tested M569 X Y S1 and M500 and indeed M501 response and all responses after that are correct 'ok'

ManuelMcLure, 2019-09-06:
There's still a bug here, but at least you have a workaround.

alexlinno, 2019-09-06:
yes, thanks for the tip

alexlinno, 2019-09-06:
sorry, it is stil a bug, shouldn't have closed it

ManuelMcLure, 2019-09-06:
From code examination, it looks like the problem is due to the unconditional call to ``CONFIG_ECHO_START()`` just after printing ``Driver stepping mode:``:
```
3286       #if HAS_STEALTHCHOP
3287         CONFIG_ECHO_HEADING("Driver stepping mode:");
3288         CONFIG_ECHO_START();
```
That prints the ``echo: `` start of line, but if the following code doesn't actually print anything (like in this case) then the ``ok`` just gets appended to that ``echo: ``.
I think this could be fixed by putting the call to ``CONFIG_ECHO_START()`` inside the ``say_M569()`` function instead.
However, this code is somewhat buggy - the ``M569`` lines for other tools/extruders (``I<index>`` or ``T<index>``) don't have ``echo: `` at the start, and the one for ``I1`` has embedded newlines instead of being all on one line.

ManuelMcLure, 2019-09-08:
Should be fixed by #15193.

ManuelMcLure, 2019-09-09:
@alexlinno Please test this with the latest code and close the issue if it is resolved.

alexlinno, 2019-09-09:
I will try this week...I'll have to save my EEPROM content and clear it  in order to reproduce it...

alexlinno, 2019-09-10:
OK tested 10/9/2019 version  M501:
Recv: echo:  M208 S0.00 W0.00 F480.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z0.00
Recv: echo:Stepper driver current:
Recv: echo:  M906 X700 Y700 Z800
Recv: echo:  M906 I1 Z800
Recv: echo:  M906 T0 E1100
Recv: 
Recv: echo:Driver stepping mode:
Recv: ok
Send: M155 S2
so yes the last stepper mode doesn't cause an echo:ok
fixed

closed 2019-09-10 by alexlinno