MarlinFirmware/Marlin #15177: [BUG] chopper_timing redefined error

Flibblebot, 2019-09-06:
### Description
Compiling latest 2.0bugfix using Einsy Rambo board with TMC2130 drivers set. Unable to compile because the chopper_timing variable is being redefined in trinamic.h file.

### Steps to Reproduce
Setup Trinamic drivers in configuration.h

**Expected behavior:** Nice, easy compile.
**Actual behavior:** Will not compile - returns error:
`In file included from sketch\src\module\stepper/indirection.h:44:0,
                 from sketch\src\module\stepper.h:221,
                 from sketch\src\module\stepper_indirection.cpp:36:
sketch\src\module\stepper/trinamic.h:52:3: error: conflicting declaration 'typedef struct chopper_timing_t chopper_timing_t'
 } chopper_timing_t;
   ^
In file included from sketch\src\module\stepper_indirection.cpp:32:0:
sketch\src\module\stepper_indirection.h:68:5: note: previous declaration as 'typedef struct chopper_timing_t chopper_timing_t'
   } chopper_timing_t;
     ^
In file included from sketch\src\module\stepper/indirection.h:44:0,
                 from sketch\src\module\stepper.h:221,
                 from sketch\src\module\stepper_indirection.cpp:36:
sketch\src\module\stepper/trinamic.h:54:35: error: redefinition of 'constexpr const chopper_timing_t chopper_timing'
 static constexpr chopper_timing_t chopper_timing = CHOPPER_TIMING;
                                   ^
In file included from sketch\src\module\stepper_indirection.cpp:32:0:
sketch\src\module\stepper_indirection.h:70:37: note: 'constexpr const chopper_timing_t chopper_timing' previously defined here
   static constexpr chopper_timing_t chopper_timing = CHOPPER_TIMING;
                                     ^`

teemuatlut, 2019-09-06:
You've done something that messed up your update.
Marlin no longer has a `src/module/stepper_indirection.h` file.

Flibblebot, 2019-09-06:
Sorry, don't know what happened there - I thought I was working off a clean instance. Obviously not.

Started again from scratch and it all works fine now.

closed 2019-09-06 by Flibblebot